coolsnowwolf/lede #12482: docker 不能拉取镜像

love-dy, 2024-09-19:
### 详细叙述

[image]
如图，连接ssh 安装 alist 是提示错误，上个版本还可以成功拉取镜像，请教大佬，这是我的网络设置问题吗，所有的设置都是默认，设置了passwall，上网一切正常，就是docker用不了

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r4s

### 详细日志

none

love-dy, 2024-09-19:
是公网的80 和 443 端口被 运营商封了造成的吗

love-dy, 2024-09-19:
又在x86的二级路由上试了试，docker可以拉取镜像，这应该是网路设置的问题，二级路由没有ipv6，单臂，r4s 主路由有ipv6，找来找去也没找到问题在哪 r4s装docker就是问题多，以前的版本没怎么设置直接就可以用，现在不行了

love-dy, 2024-09-19:
[image]

love-dy, 2024-09-19:
> > 又在x86的二级路由上试了试，docker可以拉取镜像，这应该是网路设置的问题，二级路由没有ipv6，单臂，r4s 主路由有ipv6，找来找去也没找到问题在哪 r4s装docker就是问题多，以前的版本没怎么设置直接就可以用，现在不行了
> > 大佬，可不可以指点一下，没有docker已经影响正常生活了
> 
> 自己搜索docker禁用ipv6

我也怀疑和ipv6有关，不过都用lean大佬的源码好几年了，以前也是同样的设置，主路由用docker没问题呀，实在是没发现现在的ipv6和以前的有什么区别，好像上一个版本515内核的时候还好好的，家里的主路由，不好随便折腾，等没人用的时候关ipv6试试

binge8, 2024-09-19:
> > > 又在x86的二级路由上试了试，docker可以拉取镜像，这应该是网路设置的问题，二级路由没有ipv6，单臂，r4s 主路由有ipv6，找来找去也没找到问题在哪 r4s装docker就是问题多，以前的版本没怎么设置直接就可以用，现在不行了
> > > 大佬，可不可以指点一下，没有docker已经影响正常生活了
> > 
> > 
> > 自己搜索docker禁用ipv6
> 
> 我也怀疑和ipv6有关，不过都用lean大佬的源码好几年了，以前也是同样的设置，主路由用docker没问题呀，实在是没发现现在的ipv6和以前的有什么区别，好像上一个版本515内核的时候还好好的，家里的主路由，不好随便折腾，等没人用的时候关ipv6试试

科学上网全局试试吧

rrkai, 2024-09-20:
设置镜像站呗~

love-dy, 2024-09-20:
> 设置镜像站呗~

开着科学上网不需要吧，二级路由拉取镜像没问题，看错误提示好像和ipv6有关，问题最有可能是在主路由拨号这一块，瞎猜的，还没试，哈

binge8, 2024-09-20:
光靠猜你来这里问什么，自己猜着解决更好

love-dy, 2024-09-21:
> 光靠猜你来这里问什么，自己猜着解决更好

我猜你也解决不了，还是让它老老实实做主路由吧，我又编译一次，docker去掉了，主路由有什么问题会影响家庭网络

binge8, 2024-09-21:
人家在帮你出招想办法，你在这猜这不行猜那不行的，其实那两招都能解决，就不知道试试，你还在这问什么

love-dy, 2024-09-21:
> 人家在帮你出招想办法，你在这猜这不行猜那不行的，其实那两招都能解决，就不知道试试，你还在这问什么

有道理，为你点赞

love-dy, 2024-09-21:
> 您好，passwall目前無法與docker共存，如共存會導致路由器本機代理ipv4和v6，以及局域網下的ipv6代理出現問題。目前還沒有什麼好的解決辦法，如果您非得要讓路由器上passwall和docker共存，只能在passwall選擇關閉路由器本機代理來保證局域網代理正常。

非常感谢大佬的解答，我疑惑的是月初的时候passwall和docker都编译进去了，还安装了alist容器，小雅，主路由上的docker容器能正常安装和运行，lean更换6.6内核后重新编译的版本docker就不能正常运行了

coolsnowwolf, 2026-09-05:
已核对截图和当前实现。截图中的真实错误是：

```text
Get "https://registry-1.docker.io/v2/": read tcp [2408:...]:443: read: connection reset by peer
```

也就是 dockerd 在路由器本机解析 Docker Hub 后走 IPv6 直连，连接在外部链路被重置；不是 Docker bridge 防火墙转发失败，也没有证据指向 6.6 内核。启用 Passwall 后，路由器本机流量是否代理、IPv6 是否代理以及 DNS 返回结果都会影响这里，因此旁路由无 IPv6 能成功并不能证明是内核问题。

当前仓库的 dockerd 已支持 `list registry_mirrors`、`list dns`、`option http_proxy` / `https_proxy` 以及 `option ipv6`；Dockerman 当前实现也会直接流式显示 Docker API 返回的错误，无需再改代码。可按实际网络任选其一处理：

1. 在 Passwall 中让“本机代理”覆盖 Docker Hub 的 IPv4/IPv6；
2. 在 `/etc/config/dockerd` 配置可用的 `https_proxy` 或 `registry_mirrors` 后重启 dockerd；
3. 若运营商 IPv6 到 Docker Hub 不可用，先修复 IPv6 出口，或让 Docker Hub 域名只走可用的 IPv4 路径。

定位时可分别执行 `curl -4 -I https://registry-1.docker.io/v2/`、`curl -6 -I https://registry-1.docker.io/v2/` 和 `docker pull hello-world`；前两者的差异即可确认链路。原报告没有完整系统日志，且截图已经给出外部 IPv6 reset，按网络/第三方代理配置问题关闭。

closed 2026-09-05 by coolsnowwolf