coolsnowwolf/lede #12487: rax3000m 编译报错

1967937965, 2024-09-20:
### 详细叙述

patching file Documentation/devicetree/bindings/net/pcs/mediatek,sgmiisys.yaml

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/248-v6.8-clk-mediatek-add-pcw_chg_bit-control-for-PLLs-of-MT7.patch using plaintext: 
patching file drivers/clk/mediatek/clk-pll.c
patching file drivers/clk/mediatek/clk-pll.h

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/249-v6.8-clk-mediatek-add-drivers-for-MT7988-SoC.patch using plaintext: 
patching file drivers/clk/mediatek/Kconfig
patching file drivers/clk/mediatek/Makefile
patching file drivers/clk/mediatek/clk-mt7988-apmixed.c
patching file drivers/clk/mediatek/clk-mt7988-eth.c
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c
patching file drivers/clk/mediatek/clk-mt7988-topckgen.c
patching file drivers/clk/mediatek/clk-mt7988-xfipll.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/250-clk-mediatek-add-infracfg-reset-controller-for-mt798.patch using plaintext: 
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/250-dt-bindings-reset-mediatek-add-MT7988-reset-IDs.patch using plaintext: 
patching file include/dt-bindings/reset/mediatek,mt7988-resets.h

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/251-v6.8-watchdog-mediatek-mt7988-add-wdt-support.patch using plaintext: 
patching file drivers/watchdog/mtk_wdt.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/252-clk-mediatek-mt7988-infracfg-fix-clocks-for-2nd-PCIe.patch using plaintext: 
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/253-pinctrl-mediatek-mt7981-add-additional-uart-group.patch using plaintext: 
patching file drivers/pinctrl/mediatek/pinctrl-mt7981.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/254-pinctrl-mediatek-mt7981-add-additional-emmc-group.patch using plaintext: 
patching file drivers/pinctrl/mediatek/pinctrl-mt7981.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/320-v6.2-mmc-mediatek-add-support-for-MT7986-SoC.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/321-v6.2-mmc-mtk-sd-add-Inline-Crypto-Engine-clock-control.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #4 succeeded at 2669 (offset -4 lines).

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/322-v6.2-mmc-mtk-sd-fix-two-spelling-mistakes-in-comment.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #2 succeeded at 1227 with fuzz 1 (offset -2 lines).

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/323-v6.2-mmc-Avoid-open-coding-by-using-mmc_op_tuning.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #1 FAILED at 1224.
Hunk #2 FAILED at 1320.
2 out of 2 hunks FAILED -- saving rejects to file drivers/mmc/host/mtk-sd.c.rej
Patch failed!  Please fix /home/newuser/lede/target/linux/mediatek/patches-6.1/323-v6.2-mmc-Avoid-open-coding-by-using-mmc_op_tuning.patch!
make[4]: *** [Makefile:29: /home/newuser/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.111/.prepared_993540f831a41e9e66a5e34219601c63] Error 1
make[4]: Leaving directory '/home/newuser/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/newuser/lede/target/linux'
time: target/linux/compile#11.65#7.84#14.79
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/newuser/lede'
make[1]: *** [target/Makefile:23: /home/newuser/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/newuser/lede'
make: *** [/home/newuser/lede/include/toplevel.mk:231: world] Error 2
newuser@RainYun-m2ZNrwu1:~/lede$ 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rax3000m NAND

### 详细日志

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/246-v6.8-dt-bindings-clock-mediatek-add-MT7988-clock-IDs.patch using plaintext: 
patching file include/dt-bindings/clock/mediatek,mt7988-clk.h

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/247-v6.8-dt-bindings-clock-mediatek-add-clock-controllers-of-.patch using plaintext: 
patching file Documentation/devicetree/bindings/arm/mediatek/mediatek,infracfg.yaml
patching file Documentation/devicetree/bindings/clock/mediatek,apmixedsys.yaml
patching file Documentation/devicetree/bindings/clock/mediatek,ethsys.yaml
patching file Documentation/devicetree/bindings/clock/mediatek,mt7988-ethwarp.yaml
patching file Documentation/devicetree/bindings/clock/mediatek,mt7988-xfi-pll.yaml
patching file Documentation/devicetree/bindings/clock/mediatek,topckgen.yaml
patching file Documentation/devicetree/bindings/net/pcs/mediatek,sgmiisys.yaml

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/248-v6.8-clk-mediatek-add-pcw_chg_bit-control-for-PLLs-of-MT7.patch using plaintext: 
patching file drivers/clk/mediatek/clk-pll.c
patching file drivers/clk/mediatek/clk-pll.h

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/249-v6.8-clk-mediatek-add-drivers-for-MT7988-SoC.patch using plaintext: 
patching file drivers/clk/mediatek/Kconfig
patching file drivers/clk/mediatek/Makefile
patching file drivers/clk/mediatek/clk-mt7988-apmixed.c
patching file drivers/clk/mediatek/clk-mt7988-eth.c
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c
patching file drivers/clk/mediatek/clk-mt7988-topckgen.c
patching file drivers/clk/mediatek/clk-mt7988-xfipll.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/250-clk-mediatek-add-infracfg-reset-controller-for-mt798.patch using plaintext: 
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/250-dt-bindings-reset-mediatek-add-MT7988-reset-IDs.patch using plaintext: 
patching file include/dt-bindings/reset/mediatek,mt7988-resets.h

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/251-v6.8-watchdog-mediatek-mt7988-add-wdt-support.patch using plaintext: 
patching file drivers/watchdog/mtk_wdt.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/252-clk-mediatek-mt7988-infracfg-fix-clocks-for-2nd-PCIe.patch using plaintext: 
patching file drivers/clk/mediatek/clk-mt7988-infracfg.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/253-pinctrl-mediatek-mt7981-add-additional-uart-group.patch using plaintext: 
patching file drivers/pinctrl/mediatek/pinctrl-mt7981.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/254-pinctrl-mediatek-mt7981-add-additional-emmc-group.patch using plaintext: 
patching file drivers/pinctrl/mediatek/pinctrl-mt7981.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/320-v6.2-mmc-mediatek-add-support-for-MT7986-SoC.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/321-v6.2-mmc-mtk-sd-add-Inline-Crypto-Engine-clock-control.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #4 succeeded at 2669 (offset -4 lines).

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/322-v6.2-mmc-mtk-sd-fix-two-spelling-mistakes-in-comment.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #2 succeeded at 1227 with fuzz 1 (offset -2 lines).

Applying /home/newuser/lede/target/linux/mediatek/patches-6.1/323-v6.2-mmc-Avoid-open-coding-by-using-mmc_op_tuning.patch using plaintext: 
patching file drivers/mmc/host/mtk-sd.c
Hunk #1 FAILED at 1224.
Hunk #2 FAILED at 1320.
2 out of 2 hunks FAILED -- saving rejects to file drivers/mmc/host/mtk-sd.c.rej
Patch failed!  Please fix /home/newuser/lede/target/linux/mediatek/patches-6.1/323-v6.2-mmc-Avoid-open-coding-by-using-mmc_op_tuning.patch!
make[4]: *** [Makefile:29: /home/newuser/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.111/.prepared_993540f831a41e9e66a5e34219601c63] Error 1
make[4]: Leaving directory '/home/newuser/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/newuser/lede/target/linux'
time: target/linux/compile#11.15#7.48#14.27
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/newuser/lede'
make[1]: *** [target/Makefile:23: /home/newuser/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/newuser/lede'
make: *** [/home/newuser/lede/include/toplevel.mk:231: world] Error 2
newuser@RainYun-m2ZNrwu1:~/lede$

Nicepaul, 2024-09-21:
也是报这个错

WongKwongYu, 2024-09-21:
一样

closed 2024-09-21 by aiamadeus

gforce1206, 2024-09-25:
+1，mt3000

commit 801f3451 (2024-09-21): kernel: refresh 6.1 patches

Fixes: 718ac37 ("kernel: bump 6.1 to 6.1.111")
Closed: #12487