coolsnowwolf/lede #12488: 树莓派4编译有问题

rex5652, 2024-09-22:
### 详细叙述

950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch
好像是这个补丁出错了

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Raspberry Pi4

### 详细日志

Applying /home/runner/work/OpenWrt/OpenWrt/openwrt/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch using plaintext: 
patching file net/bluetooth/smp.c
Hunk #1 FAILED at 2214.
Hunk #2 succeeded at 2223 (offset -6 lines).
1 out of 2 hunks FAILED -- saving rejects to file net/bluetooth/smp.c.rej
Patch failed!  Please fix /home/runner/work/OpenWrt/OpenWrt/openwrt/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch!
make[4]: *** [Makefile:43: /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.111/.prepared_7290c947021f75253a8026e9d6ed1837] Error 1

yz-java, 2024-09-23:
将这个文件 lede/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch 内容替换

From 54495a03bc79fbf340cc5657ab9a15d26798d7e1 Mon Sep 17 00:00:00 2001
From: yangzhao <yangzhaojava@gmail.com>
Date: Tue, 24 Sep 2024 03:13:21 +0800
Subject: [PATCH] Revert "Bluetooth: Always request for user
 confirmation for Just Works (LE SC)"

---
 net/bluetooth/smp.c | 5 +----
 1 file changed, 1 insertion(+), 4 deletions(-)

diff --git a/net/bluetooth/smp.c b/net/bluetooth/smp.c
index b93494790..c26b4fa32 100644
--- a/net/bluetooth/smp.c
+++ b/net/bluetooth/smp.c
@@ -2208,7 +2208,7 @@ static u8 smp_cmd_pairing_random(struct l2cap_conn *conn, struct sk_buff *skb)
 	if (err)
 		return SMP_UNSPECIFIED;
 
-	if (smp->method == REQ_OOB) {
+	if (smp->method == JUST_WORKS || smp->method == REQ_OOB) {
 		if (test_bit(SMP_FLAG_INITIATOR, &smp->flags)) {
 			sc_dhkey_check(smp);
 			SMP_ALLOW_CMD(smp, SMP_CMD_DHKEY_CHECK);
@@ -2223,9 +2223,6 @@ static u8 smp_cmd_pairing_random(struct l2cap_conn *conn, struct sk_buff *skb)
 	confirm_hint = 0;
 
 confirm:
-	if (smp->method == JUST_WORKS)
-		confirm_hint = 1;
-
 	err = mgmt_user_confirm_request(hcon->hdev, &hcon->dst, hcon->type,
 					hcon->dst_type, passkey, confirm_hint);
 	if (err)
-- 
2.43.4

yz-java, 2024-09-23:
将lede/target/linux/bcm27xx/patches-6.1/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.patch 文件替换txt文件里的文本
[950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.txt](https://github.com/user-attachments/files/17103483/950-0078-Revert-Bluetooth-Always-request-for-user-confirmatio.txt)

coolsnowwolf, 2024-09-29:
已修复

closed 2024-09-29 by coolsnowwolf