coolsnowwolf/lede #12491: 友善派 新的板子 nanopi r3s 出来了什么时候支持一下

1021683053, 2024-09-24:
### 详细叙述

https://www.friendlyelec.com/index.php?route=product/product&product_id=303

https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R3S

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Nanopi r3s

### 详细日志

无此型号

iKira, 2024-10-28:
2024-10-22 12:09:10

今天的源代码已更新：

1. 为友善 R6C/R6S/Nano PC T6 系列提供主线支持（支持 GPU等）。**下一波开源 R3S~**
2. 支持各种 Rockchip 新的芯片，包括 RK3528/RK3582 等
3. 支持国产 WIFI 芯片，例如 AIC8800 的 WIFI6 AP 模式（才 10块钱的芯片）
4. 更新优化一大波 LUCI JS 23.05 的 界面和 APP

Enjoy~

TaneikaMacedo, 2024-11-14:
18 sec ago Mujib Zalmay ʟᴇᴀᴋᴇᴅ ᴠɪᴅᴇᴏ Viral On Social Media. ◉Free(✧VIDEO𝕊✧ＸＸＸ✧𝙵𝚞𝚕𝚕 ORIGINAL)◉ Mujib Zalmay Viral ᴠɪᴅᴇᴏ ʟᴇᴀᴋᴇᴅ Update LINK On Social Media 𝕏𝘛𝘸𝘪𝘵𝘵𝘦𝘳

[𝑾𝒂𝒕𝒄𝒉 🟢 ➤ ➤ ➤ 🌐 𝖢𝗅𝗂𝖼𝗄 𝖧𝖾𝗋𝖾 𝖳𝗈 𝗅𝗂𝗇𝗄 (𝖥𝗎𝗅 𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇𝗄)](https://nowviral.pages.dev/?vidzm1)

[🔴 ➤► 𝖣𝖮𝖶𝖭𝖫𝖮𝖠𝖣👉👉 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇 𝗄) 𝖶𝖺𝗍𝖼𝗁](https://nowviral.pages.dev/?vidzm2)

[🟢 ➤ ➤ ➤ 🌐 𝖢𝗅𝗂𝖼𝗄 𝖧𝖾𝗋𝖾 𝖳𝗈 𝗅𝗂𝗇𝗄 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺 𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇𝗄)](https://nowviral.pages.dev/?vidzm3)

[🔴 ➤► 𝖣𝖮𝖶𝖭𝖫𝖮𝖠𝖣👉👉 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇 𝗄)](https://nowviral.pages.dev/?vidzm4)

Mujib Zalmay's latest video has taken the internet by storm, sparking a wave of excitement and discussion across social media platforms. The rising digital influencer, known for her captivating and creative content—especially her recent Spiderman-themed video—has experienced an explosive surge in popularity. Fans are eagerly awaiting the full version, while viewers flock to Instagram, X (formerly Twitter), and Discord to share reactions and engage with her latest masterpiece.

The overwhelming buzz surrounding the video highlights the massive role social media plays in propelling creators like Mujib Zalmay into the limelight. With her fresh approach and unique style, she has not only captured attention but also inspired countless fans. This viral success serves as a powerful reminder of how social media can amplify visibility and deepen the connection between creators and their audiences.

With such a major hit on her hands, fans are left eagerly anticipating what Mujib has in store next—knowing it will undoubtedly be another game-changer!
.

.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.

10 sec ago Mujib Zalmay ʟᴇᴀᴋᴇᴅ ᴠɪᴅᴇᴏ Viral On Social Media. ★(ＮＯＷ~𝐅𝐔𝐋𝐋*:.VIDEO𝕊) Mujib Zalmay Viral ᴠɪᴅᴇᴏ ʟᴇᴀᴋᴇᴅ Update Files Trending On Social Media X 𝕏𝘛𝘸𝘪𝘵𝘵𝘦𝘳

[𝑾𝒂𝒕𝒄𝒉 🟢 ➤ ➤ ➤ 🌐 𝖢𝗅𝗂𝖼𝗄 𝖧𝖾𝗋𝖾 𝖳𝗈 𝗅𝗂𝗇𝗄 (𝖥𝗎𝗅 𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇𝗄)](https://nowviral.pages.dev/?vidzm1)

[🔴 ➤► 𝖣𝖮𝖶𝖭𝖫𝖮𝖠𝖣👉👉 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇 𝗄) 𝖶𝖺𝗍𝖼𝗁](https://nowviral.pages.dev/?vidzm2)

[🟢 ➤ ➤ ➤ 🌐 𝖢𝗅𝗂𝖼𝗄 𝖧𝖾𝗋𝖾 𝖳𝗈 𝗅𝗂𝗇𝗄 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺 𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇𝗄)](https://nowviral.pages.dev/?vidzm3)

[🔴 ➤► 𝖣𝖮𝖶𝖭𝖫𝖮𝖠𝖣👉👉 (𝖥𝗎𝗅𝗅 𝖵𝗂𝗋𝖺𝗅 𝖵𝗂𝖽𝖾𝗈 𝖫𝗂𝗇 𝗄)](https://nowviral.pages.dev/?vidzm4)

Mujib Zalmay's latest video has taken the internet by storm, sparking a wave of excitement and discussion across social media platforms. The rising digital influencer, known for her captivating and creative content—especially her recent Spiderman-themed video—has experienced an explosive surge in popularity. Fans are eagerly awaiting the full version, while viewers flock to Instagram, X (formerly Twitter), and Discord to share reactions and engage with her latest masterpiece.

The overwhelming buzz surrounding the video highlights the massive role social media plays in propelling creators like Mujib Zalmay into the limelight. With her fresh approach and unique style, she has not only captured attention but also inspired countless fans. This viral success serves as a powerful reminder of how social media can amplify visibility and deepen the connection between creators and their audiences.

With such a major hit on her hands, fans are left eagerly anticipating what Mujib has in store next—knowing it will undoubtedly be another game-changer!
.

.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.
.

commit f21df973 (2024-12-23): uboot-rockchip: fix NanoPi-R3S defconfig (#13232)

Fixes: #12491, #13001
closed 2024-12-23 by aiamadeus