coolsnowwolf/lede #12493: H28K无法启动

Gabrielxzx, 2024-09-25:
### 详细叙述

如题，看到最近支持了H28K，编译了新系统后，烧录到EMMC后却无法启动，尝试把固件烧录到SD卡后发现可以正常启动。但这样就有点浪费EMMC了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

HINLINK OPC-H28K Board

### 详细日志

研究了下板子，没发现有预留UART接口，暂时没日志。
看了下用的物料：
EMMC：KLM8G1GETF-B041
不知道是不是兼容问题

aiamadeus, 2024-09-26:
你 revert 到 3aad94cd07a40d3d4c1b8cecabefa807fd0b8149 试一下
另外 rkbin 有毒，刷机用的 loader 需要用最新的

Gabrielxzx, 2024-09-26:
> 你回退到 [3aad94c](https://github.com/coolsnowwolf/lede/commit/3aad94cd07a40d3d4c1b8cecabefa807fd0b8149) 试一下 另外 rkbin 有毒，刷机用的 loader 需要用最新的

问下这个最新的loader哪里能获得？

Gabrielxzx, 2024-09-26:
> 你回退到 [3aad94c](https://github.com/coolsnowwolf/lede/commit/3aad94cd07a40d3d4c1b8cecabefa807fd0b8149) 试一下 另外 rkbin 有毒，刷机用的 loader 需要用最新的

用瑞芯微开发工具擦除一遍Flash就可以启动了

[image]

closed 2024-09-26 by Gabrielxzx