coolsnowwolf/lede #12503: 编译时加入kenzok8/small-package中的qbit会报错,且如果取消qbit编译成功后从istore下载qbit,aiar2,Transmission全部失败

xiaojiejieouo, 2024-09-28:
### 详细叙述

如题,所以想提问,如何能使用bt下载的软件?需要我加一些什么吗还是做如何改变
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

errors:
 * pkg_hash_check_unresolved: cannot find dependency qbittorrent for luci-app-qbittorrent
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-qbittorrent found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-qbittorrent.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-qbittorrent-zh-cn:
 * 	qbittorrent
 * opkg_install_cmd: Cannot install package luci-i18n-qbittorrent-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/xiaojiejieouo/lede'
make[1]: *** [package/Makefile:111: /home/xiaojiejieouo/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xiaojiejieouo/lede'
make: *** [/home/xiaojiejieouo/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2024-09-29:
?

xiaojiejieouo, 2024-09-29:
> ?
我看([aiamadeus](https://github.com/aiamadeus))更改了我的标题,他好像误解了我的意思
就是大概数个月前我使用您的源代码编译出来的openwrt可以使用istore商店中的aiar2,qbit等BT下载软件,但是前些天我编译时加入kenzok8/small-package中的qbit会报错,报错代码如详细日志中显示,如果编译时取消qbit则可以正常编译,然后从istore商店中下载qbit,或者aiar2都无法下载,如截图所示
所以我的目的是,如何在用您的源码编译出的openwrt中使用这些bt软件?是我需要更改什么设置吗,如果是其他软件的问题,如有打扰抱歉

xiaojiejieouo, 2024-09-29:
在尝试启动qbit时会显示
WebUI will be started shortly after internal preparations. Please wait...
terminate called after throwing an instance of 'boost::system::system_error'
  what():  context: dso not found (engine routines) [asio.ssl:318767236]
Aborted

coolsnowwolf, 2024-09-29:
你来错地方了，去小鸡过河那边问问？

xiaojiejieouo, 2024-09-30:
> 你来错地方了，去小鸡过河那边问问？

好的抱歉,谢谢大佬了

closed 2024-11-20 by coolsnowwolf