coolsnowwolf/lede #12508: 最近一次package里面的zerotier的pull request commit好像有问题

GreatMichaelLee, 2024-09-30:
### 详细叙述

如题，https://github.com/coolsnowwolf/packages/pull/857 这个commit提交后luci里面的networkid和nat配置就显示不出来了,见图示。

没有合入这个commit之前的luci界面。
[image]

合入之后，启用，networkid和使能NAT三个配置项均没有了，是不是这个改动跟现有的luci不适配？如是的话，哪位大佬能改下Luci?
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无，见详细叙述。

zuni11, 2024-10-01:
同样的问题，NAT设置失效了，都不通啦

XuHuapeng001, 2024-10-02:
一样的问题，我还以为是我编译有什么问题导致的，把编译环境重新安装了一遍，还是一样的结果。
其他第三方固件群里也有同样的反馈。

oldmmry, 2024-10-03:
并且加了zerotier编译出来的版本会导致启动项web界面无法打开。

GreatMichaelLee, 2024-10-04:
> 看这个https://github.com/mwarning/zerotier-openwrt/wiki/Configure-ZeroTier-on-OpenWrt

你说的这个article和这个issue有什么关系呢？没看明白，这个article讲的无非是zerotier在Openwrt的全量配置项。我提的是Luci在新的commit之后显示不全的问题。是新加了什么配置项吗

GreatMichaelLee, 2024-10-04:
> > > 看这个https://github.com/mwarning/zerotier-openwrt/wiki/Configure-ZeroTier-on-OpenWrt
> > 
> > 
> > 你说的这个article和这个issue有什么关系呢？没看明白，这个article讲的无非是zerotier在Openwrt的全量配置项。我提的是Luci在新的commit之后显示不全的问题。是新加了什么配置项吗
> 
> 之前配置文件，启用和加入网络是在sample_config 下。现在配置文件，启用在global下，并且加入network 默认是earth。而lean的zerotier luci 界面就是直接修改的sample_config。
> 
> 1.14.1 root@OpenWrt:~# uci show zerotier zerotier.global=zerotier zerotier.global.enabled='0' zerotier.earth=network zerotier.earth.id='8056c2e21c000001' zerotier.earth.allow_managed='1' zerotier.earth.allow_global='0' zerotier.earth.allow_default='0' zerotier.earth.allow_dns='0'
> 
> 1.14.1之前的 root@OpenWrt:~# uci show zerotier zerotier.sample_config=zerotier zerotier.sample_config.enabled='0' zerotier.sample_config.join='8056c2e21c000001'

我比较了一下，确实是你说的这个问题，感谢指教，我现在两边两个异地路由器跑的两个版本，路由器本身互通可以，但路由器后LAN的NAT挂了，不通。
麻烦大佬修一下啊。

zuni11, 2024-10-04:
临时解决方案，防火墙的地方把转发打开。

zuni11, 2024-10-07:
L大回滚了，但是重新编译保留配置的话并不会恢复luci，因为配置文件不对了，可以修改/etc/config/zerotier文件，改成下面的内容，然后重启就全恢复了。
config zerotier 'sample_config'
	option enabled '1'
	list join 'xxxxxxxxxxxxxxxxxxx'  这里写你自己的id
	option nat '1'

rshi0212, 2024-11-02:
帮大忙了 昨天编译了新的 就发现不对劲 乱成一锅粥

> L大回滚了，但是重新编译保留配置的话并不会恢复luci，因为配置文件不对了，可以修改/etc/config/zerotier文件，改成下面的内容，然后重启就全恢复了。 config zerotier 'sample_config' option enabled '1' list join 'xxxxxxxxxxxxxxxxxxx' 这里写你自己的id option nat '1'

rshi0212, 2024-11-02:
> L大回滚了，但是重新编译保留配置的话并不会恢复luci，因为配置文件不对了，可以修改/etc/config/zerotier文件，改成下面的内容，然后重启就全恢复了。 config zerotier 'sample_config' option enabled '1' list join 'xxxxxxxxxxxxxxxxxxx' 这里写你自己的id option nat '1'

没搞定，配置文件没问题，整个机器重置也无法互相访问，显示拒绝连接

rshi0212, 2024-11-03:
> > L大回滚了，但是重新编译保留配置的话并不会恢复luci，因为配置文件不对了，可以修改/etc/config/zerotier文件，改成下面的内容，然后重启就全恢复了。 config zerotier 'sample_config' option enabled '1' list join 'xxxxxxxxxxxxxxxxxxx' 这里写你自己的id option nat '1'
> 
> 没搞定，配置文件没问题，整个机器重置也无法互相访问，显示拒绝连接

搞定了，A/B两端都升级了下，然后按照https://github.com/mwarning/zerotier-openwrt/wiki走了一遍。

closed 2026-09-05 by coolsnowwolf