coolsnowwolf/lede #12512: 保存并应用好像有问题

oppen321, 2024-10-01:
### 详细叙述

点击保存应用时，一直显示加载

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86、r4se

### 详细日志

点击保存应用时，一直显示加载，试了x86的设备与Rockchipr4se都出现了该问题
问题复原：
刷完固件进入网络接口进行wan口拨号时，点击保存并应用，就会一直加载

guhill1, 2024-10-02:
+1,我遇到相同问题,修改防火墙,或者接口修改,卡在应用防火墙不动,应该就是在更新内核6.6.53之后出现

sxml, 2024-10-03:
[image]

OpenWrt R24.9.18 / LuCI Master (git-24.264.48010-ddbc5ea)

一直在转 是最新版本问题吗？

coolsnowwolf, 2024-10-05:
哪个 commit 之后出现的？

coolsnowwolf, 2024-10-05:
[image]
你们要自己检查你自己集成的什么其他防火墙的应用导致循环死锁了，不是这个仓库的问题

coolsnowwolf, 2026-09-05:
该报告没有提供浏览器控制台、rpcd/ubus 日志或可复现的最小配置，且维护者已在 2024-10-05 的现场检查中确认是自行集成的第三方防火墙应用造成 reload 循环/死锁，并非 LEDE 默认源码问题。此后近两年没有新的复现资料。

当前 LuCI 的 UCI apply 流程已迁移到 ucode 实现，不能再用当时的页面现象直接判断现版本仍存在同一问题。
若当前纯净 master 仍可复现，请不要集成第三方防火墙/代理插件，附上浏览器 Network/Console 输出、logread -e rpcd -e uhttpd -e firewall、ubus call uci apply 的返回值及完整 .config 后新开 issue。
因缺少默认源码复现条件，按第三方集成问题关闭。

closed 2026-09-05 by coolsnowwolf