coolsnowwolf/lede #12513: 新装的Ubuntu 24.04.1 LTS 无法安装环境

chiyu521, 2024-10-02:
### 详细叙述

chiyu@chiyu-Home:~$ sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev \
libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树... 完成
正在读取状态信息... 完成                 
注意，选中 'libncurses-dev' 而非 'libncurses5-dev'
注意，选中 'libncurses-dev' 而非 'libncursesw5-dev'
注意，选中 'genisoimage' 而非 'mkisofs'
E: 无法定位软件包 python2.7
E: 无法按照 glob ‘python2.7’ 找到任何软件包

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Ubuntu 24.04.1 LTS

### 详细日志

无

oppen321, 2024-10-02:
不是写了推荐用debian11和ubuntu22.04吗

chiyu521, 2024-10-02:
官网最低是23.10 本来想重装系统 顺便换最新的了

oppen321, 2024-10-02:
你可以去这里https://cdimage.debian.org/cdimage/archive/下载debian11的镜像

hzmax0416, 2024-10-02:
> 官网最低是23.10 本来想重装系统 顺便换最新的了

官网有22.04版本啊

hcym, 2024-10-02:
刚刚也是全新的，少几个出错的没什么问题
编译的插件多的话还是用22稳妥

xingshu1990, 2024-10-04:
国庆期间刚编译了树莓派3B+的  基于LEDE的openwrt固件，虽然编译出来后的挂载点可能还有问题。

Beginner-Go, 2024-10-04:
自己想办法装 `python2.7`

fireflyddt, 2024-10-05:
> 官网最低是23.10 本来想重装系统 顺便换最新的了

国内的话 https://mirrors.tuna.tsinghua.edu.cn/ubuntu-releases/ 清华源 可以千兆满速下载

xingshu1990, 2024-10-05:
> > 官网最低是23.10 本来想重装系统 顺便换最新的了
> 
> 国内的话 https://mirrors.tuna.tsinghua.edu.cn/ubuntu-releases/ 清华源 可以千兆满速下载

ubuntu丢给迅雷，满速下载。
相对于ubuntu来说，编译过程中的dl库的下载才叫折磨，除非没有洁癖，事先下载DL库。

l4326769, 2024-10-05:
不装插件的话debian11的编译器够用，装插件的话还是ubuntu2204吧，debian12有三个包装不上

qingtian110, 2024-10-05:
ubuntu 24.04: https://github.com/sbwml/actions/blob/openwrt-build-setup/action.yml#L12

hcym, 2024-10-27:
我也退到2204.5，不然clan13搞不定，不能编译daed呆鹅

hgq5, 2024-10-28:
我用的就是24.04.01版本，并未有你的这个情况，不同的地方是我直接安装24.04.01，并不是升级上来的。

closed 2024-11-20 by coolsnowwolf

Michaol, 2024-11-29:
24.04编译只需要以下这些包：build-essential clang flex bison g++ gawk gcc-multilib g++-multilib gettext git libncurses5-dev libssl-dev python3-setuptools rsync swig unzip zlib1g-dev file wget

lj267lj267, 2024-12-01:
> 不是写了推荐用debian11和ubuntu22.04吗

推荐没有指定版本吧？

evenwangcn, 2024-12-09:
> ubuntu 24.04: https://github.com/sbwml/actions/blob/openwrt-build-setup/action.yml#L12

CI的话，添加以下这行添加缺失的源
sudo bash -c "curl -skL https://raw.githubusercontent.com/sbwml/actions/openwrt-build-setup/sources-24.04.list > /etc/apt/sources.list"

d975xbx, 2024-12-10:
debain12,ubuntu24.04都正常编译

liwenjie119, 2024-12-21:
现在action应该是更新到24.04了吧？安装依赖会报错tex-common
Building format(s) --all.
	This may take some time... 
fmtutil failed. Output has been stored in
/tmp/fmtutil.QHcamtV1
Please include this file if you report a bug.

dpkg: error processing package tex-common (--configure):
 installed tex-common package post-installation script subprocess returned error exit status 1

vincejv, 2025-01-03:
Github CI is failing under Ubuntu 24
[image]

hgq5, 2025-01-03:
这问题都关闭了，为啥还继续讨论？？

xm0625, 2025-01-26:
建议直接Docker，省时省力，环境统一