coolsnowwolf/lede #12514: 求助make menuconfig默认配置的已选太多，导致编译固件过大。

kygsmsc, 2024-10-02:
### 详细叙述

翻出家里一台老路由，只有16M闪存，用L大源码编译自己什么都没有选择，就只是改变成路由器对应的7620A和路由器型号，结果编译出来的固件大于16M，怎么修改make menuconfig 已选配置的默认值？？？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

hc5761

### 详细日志

无

Cathgao, 2024-10-02:
16m不再支持，垃圾应该在垃圾桶里

kygsmsc, 2024-10-03:
> 16m不再支持，垃圾应该在垃圾桶里

官方openwrt最新版本，默认编译才刚6m过点。不支持有些说不过去了。

Cathgao, 2024-10-03:
那你可以用官方的啊，这个库不就是为了跟官方不一样吗？

coolsnowwolf, 2024-10-04:
看看选择了什么 golang 一类的程序导致大于 15MB 了

drw2005, 2024-10-16:
想知道哪些可以去掉的？
有没有哪位大佬来个可以去掉而不影响功能的列表？

coolsnowwolf, 2026-09-05:
这不是 HC5761 设备定义或镜像大小检查失效。当前设备定义明确设置 IMAGE_SIZE=15808k，超出时构建系统会由 check-size 拒绝生成镜像；LEDE 的 router 默认配置本身还会选择 LuCI、SSR Plus、UPnP、DDNS、文件传输等扩展应用，因此不能拿 OpenWrt 官方的精简默认镜像大小直接比较。

- 若要重新从默认值开始，请先保存旧配置，然后删除/移走 .config，再执行 make menuconfig；仅再次打开 menuconfig 不会清除以前选择。
- 用 ./scripts/diffconfig.sh 可列出相对默认值额外选择的项目，重点检查 Go/Rust 程序和代理、Docker、文件系统等大包。
- 对 16MB 设备应按需要取消 luci-app-ssr-plus、luci-app-vsftpd、luci-app-filetransfer、luci-app-nlbwmon 等非必要默认应用；不能在不知道使用需求的情况下提供一份“全部可删且不影响功能”的通用列表。
Issue 未提供 .config、diffconfig 或实际超限日志，无法归因到具体软件包；按构建配置咨询关闭。若仍有可复现的默认配置超限，请附上全新 .config、./scripts/diffconfig.sh 输出和完整 make V=s 尾部日志后重新提交。

closed 2026-09-05 by coolsnowwolf