coolsnowwolf/lede #12515: 防火墙,保存应用会卡住

guhill1, 2024-10-03:
### 详细叙述

防火墙,保存应用会卡住
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

固件版本 | OpenWrt SNAPSHOT r6884+4-c54f4958d / LuCI Master (git-24.264.48010-ddbc5ea)

内核版本 | 6.6.53

xiaol1n7, 2024-10-05:
我也遇到过，passwall2无法应用更改，然后cpu某些（有时全部）核心100%占用，htop命令看了一下是netstat -tuln在吃资源，感觉是某种bug

lanruihua026, 2024-10-07:
该bug目前修复了吗？

diabolic00, 2024-10-07:
我新装之后也是，一样~很痛苦啊

coolsnowwolf, 2024-10-09:
好歹传个固件看看集成了什么

diabolic00, 2024-10-09:
System Overview,
[image]

xingshu1990, 2024-10-09:
> System Overview, ![image](https://private-user-images.githubusercontent.com/64199202/374857843-14cb07ab-3575-4c66-a749-f0445164622b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0NjAzODYsIm5iZiI6MTcyODQ2MDA4NiwicGF0aCI6Ii82NDE5OTIwMi8zNzQ4NTc4NDMtMTRjYjA3YWItMzU3NS00YzY2LWE3NDktZjA0NDUxNjQ2MjJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA5VDA3NDgwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRkNjhmN2FmYzU3YjI4M2YyNDcwYmIyN2IzOTc1MjFiMWY4OWQyY2ExM2YzNDRiYmRiYWQ2M2UyODJlODdlZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.IEXry4_mvYCBJ7iX5BO1zz7A9SVJqwfUW0y8qsiSzOQ) 就这几个谢谢L大，L大好人啊 楼上各位L大出来了，把固件发出来看看，看看到底是啥出问题了。

应该是传固件文件上来，不是传截图。

guhill1, 2024-10-10:
> 好歹传个固件看看集成了什么

大佬好,我发现好像是配置问题,我编译了最新固件,其中也引入了kenzo大佬库,我恢复出厂设置,手工重建配置,不会造成firewall卡住的问题,但是导入旧的配置,就会造成firewall保存并应用卡住,我把新旧配置用beyondcompare做了比较,
[image]
相差这几个文件.
奇怪的是我在新配置(不会导致firewall卡住)的配置里面,把luci,firewall,advanced,adguardhome,adblock这些旧的文件移入新配置也不会造成卡顿

guhill1, 2024-10-10:
大佬好,我发现好像是配置问题,我编译了最新固件,其中也引入了kenzo大佬库,我恢复出厂设置,手工重建配置,不会造成firewall卡住的问题,但是导入旧的配置,就会造成firewall保存并应用卡住,我把新旧配置用beyondcompare做了比较,
[image]
相差这几个文件.
奇怪的是我在新配置(不会导致firewall卡住)的配置里面,把luci,firewall,advanced,adguardhome,adblock这些旧的文件移入新配置也不会造成卡顿

closed 2024-10-29 by guhill1