coolsnowwolf/lede #12520: cr660x防火墙重启没反应

yidianyipie, 2024-10-05:
### 详细叙述

Hi 大佬们好，

最近两次编译新的cr660x固件发现配置修改之后，一直卡在防火墙这边
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xiaomi cr6608

### 详细日志

```python
Thu Oct  3 22:15:31 2024 kern.notice kernel: [    0.000000] Linux version 5.4.284 (runner@fv-az1534-624) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r6885-af37c0ffb)) #0 SMP Thu Oct 3 14:00:09 2024
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router CR660x
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Zone ranges:
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000]   HighMem  empty
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Early memory node ranges
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s26640 r8192 d22512 u57344 alloc=14*4096
Thu Oct  3 22:15:31 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Thu Oct  3 22:15:31 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000030
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000030
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] Memory: 250024K/262144K available (6255K kernel code, 276K rwdata, 1312K rodata, 1232K init, 235K bss, 12120K reserved, 0K cma-reserved, 0K highmem)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] NR_IRQS: 256
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.000008] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.007882] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.074052] pid_max: default: 32768 minimum: 301
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.078841] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.086041] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.096886] rcu: Hierarchical SRCU implementation.
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.102387] smp: Bringing up secondary CPUs ...
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    3.277467] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    3.277477] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    3.277490] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Thu Oct  3 22:15:31 2024 kern.info kernel: [    3.277595] CPU1 revision is: 0001992f (MIPS 1004Kc)
Thu Oct  3 22:15:31 2024 kern.info kernel: [    0.167171] Synchronize counters for CPU 1: done.
Thu Oct  3 22:15:31 2024 kern.warn kernel: [    3.370822] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 517 lines cut ...]
Sat Oct  5 10:34:22 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: disassociated
Sat Oct  5 10:34:23 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Oct  5 10:37:28 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: associated (aid 1)
Sat Oct  5 10:37:28 2024 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 3c:06:30:08:a8:08 auth_alg=sae
Sat Oct  5 10:37:28 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 RADIUS: starting accounting session 220F9BE10311AC31
Sat Oct  5 10:37:28 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 WPA: pairwise key handshake completed (RSN)
Sat Oct  5 10:37:28 2024 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 3c:06:30:08:a8:08
Sat Oct  5 10:38:43 2024 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 3c:06:30:08:a8:08
Sat Oct  5 10:38:43 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: disassociated
Sat Oct  5 10:38:45 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Oct  5 10:39:00 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: associated (aid 1)
Sat Oct  5 10:39:00 2024 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 3c:06:30:08:a8:08 auth_alg=sae
Sat Oct  5 10:39:00 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 RADIUS: starting accounting session BAC86FCE1A3C0659
Sat Oct  5 10:39:00 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 WPA: pairwise key handshake completed (RSN)
Sat Oct  5 10:39:00 2024 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 3c:06:30:08:a8:08
Sat Oct  5 10:39:23 2024 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 3c:06:30:08:a8:08
Sat Oct  5 10:39:23 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: disassociated
Sat Oct  5 10:39:24 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Oct  5 10:42:39 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 IEEE 802.11: associated (aid 1)
Sat Oct  5 10:42:39 2024 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 3c:06:30:08:a8:08 auth_alg=sae
Sat Oct  5 10:42:39 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 RADIUS: starting accounting session CC8EB26205850711
Sat Oct  5 10:42:39 2024 daemon.info hostapd: wlan1: STA 3c:06:30:08:a8:08 WPA: pairwise key handshake completed (RSN)
Sat Oct  5 10:42:39 2024 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 3c:06:30:08:a8:08

```

yidianyipie, 2024-10-05:
在terminal上面重启防火墙也是一直卡着，没有任何反应
[image]
这个有什么办法解决不，是因为编译问题，还是其他的问题？

sxml, 2024-10-05:
同样点应用 卡住 一直在转动

coolsnowwolf, 2024-10-05:
[image]
你们要自己检查你自己集成的什么其他防火墙的应用导致循环死锁了，不是这个仓库的问题

sxml, 2024-10-06:
[image]

固件版本OpenWrt R24.9.18 / LuCI Master (git-24.264.48010-ddbc5ea)
内核版本6.6.54

卡卡卡 卡住

yidianyipie, 2024-10-08:
> ![image](https://private-user-images.githubusercontent.com/31687149/373855433-fd063da9-a412-4c43-b126-f66469ef7ca1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

感谢lean大佬的回复，我这边经过多次的尝试，找到了问题
在编译固件的时候，安装的依赖里面加上了下面的几个包，试了下防火墙没再卡住了
```python
libjemalloc-dev uuid-dev libunwind-dev libsafec-dev  libflatbuffers-dev libhyperscan-dev libmnl-dev libluajit-5.1-dev nftables liblzma-dev libpcre3-dev libhwloc-dev luajit
```

closed 2024-10-08 by yidianyipie