coolsnowwolf/lede #12521: R8157 USB网卡正常识别，但插入网线10秒后崩溃

loydcheung, 2024-10-05:
### 详细叙述

X86 J4125四口主机，看到最近加入了r8157 5G USB有线网卡支持，迫不及待买来测试。
[image]
可以正常识别，显示为第五个网口eth4，并加入LAN。刚插入网线的前几秒可以用，但10秒左右就会崩溃。
这块网卡是USB3.0的，网上搜了一圈好像USB3.0有关的崩溃和假死问题很多，包括移动硬盘等等。
由于假死后整个网络崩溃也无法进入系统看日志。
后面换了r8156 2.5G网卡测试也是一样假死。
感觉是USB3.0的支持问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86 J4125

### 详细日志

假死无法进入系统截取

loydcheung, 2024-10-05:
我已经编译加入了kmod-usb3，并且按照官方文档加入了所有依赖，情况依然没有改变。[image]

aiamadeus, 2024-10-05:
你截图的这个是上游的驱动，跟螃蟹官方的 r8152 驱动（bug 多）是冲突的。

loydcheung, 2024-10-05:
> 你截图的这个是上游的驱动，跟螃蟹官方的 r8152 驱动（bug 多）是冲突的。

我实际编译的是默认选的那个带-vendor的驱动，大佬帮忙看看，还有什么可能出现问题。

loydcheung, 2024-10-08:
今天在未插网线的情况下，运行了ethtool和lsusb查看网口信息（插了网线整个系统会崩溃，只能这样）

root@OpenWrt_J4125:~# ethtool eth4
Settings for eth4:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
                                5000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: d
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: no

root@OpenWrt_J4125:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 2: Dev 4, If 0, Class=, Driver=r8152, 480M

ethtool返回的结果是正常的，表示正常识别。
但lsusb返回结果发现USB网卡实际是运行在480M的USB2.0模式下。我实际是插在USB3.0口上的。
所以应该问题还是出在USB3.0的支持上面。（我实际已经编译进了kmod-usb3）

loydcheung, 2024-10-08:
至此，已经确定驱动可以正常识别r8157这张5Gbps速率的USB网卡，并且确认是能工作在usb3.0状态。但工作不超过1分钟整个系统就会假死，（所有网口指示灯都正常闪烁但网络全部无法使用）

另外测试了另一块USB网卡r8156，也是同样的情况。

请各位大佬帮忙看看。

以下是我的lspci：
root@OpenWrt_J4125:~# lspci
00:00.0 Host bridge: Intel Corporation Gemini Lake Host Bridge (rev 06)
00:02.0 VGA compatible controller: Intel Corporation GeminiLake [UHD Graphics 600] (rev 06)
00:0e.0 Audio device: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio (rev 06)
00:0f.0 Communication controller: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface (rev 06)
00:12.0 SATA controller: Intel Corporation Celeron/Pentium Silver Processor SATA Controller (rev 06)
00:13.0 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f6)
00:13.2 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f6)
00:13.3 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f6)
00:14.0 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f6)
00:14.1 PCI bridge: Intel Corporation Gemini Lake PCI Express Root Port (rev f6)
00:15.0 USB controller: Intel Corporation Celeron/Pentium Silver Processor USB 3.0 xHCI Controller (rev 06)
00:17.0 Signal processing controller: Intel Corporation Device 31b4 (rev 06)
00:17.1 Signal processing controller: Intel Corporation Device 31b6 (rev 06)
00:17.2 Signal processing controller: Intel Corporation Device 31b8 (rev 06)
00:17.3 Signal processing controller: Intel Corporation Device 31ba (rev 06)
00:1c.0 SD Host controller: Intel Corporation Celeron/Pentium Silver Processor SDA Standard Compliant SD Host Controller (rev 06)
00:1f.0 ISA bridge: Intel Corporation Celeron/Pentium Silver Processor LPC Controller (rev 06)
00:1f.1 SMBus: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model (rev 06)
01:00.0 Non-Volatile memory controller: Intel Corporation NVMe Optane Memory Series
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)

loydcheung, 2024-10-08:
再补上我的config：

```

CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y

CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="x86"
CONFIG_TARGET_SUBTARGET="64"
CONFIG_TARGET_PROFILE="DEVICE_generic"
CONFIG_TARGET_ARCH_PACKAGES="x86_64"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe"
CONFIG_CPU_TYPE=" "
CONFIG_LINUX_6_6=y
CONFIG_DEFAULT_amd64-microcode=y
CONFIG_DEFAULT_autocore-x86=y
CONFIG_DEFAULT_automount=y
CONFIG_DEFAULT_autosamba=y
CONFIG_DEFAULT_base-files=y
CONFIG_DEFAULT_block-mount=y
CONFIG_DEFAULT_busybox=y
CONFIG_DEFAULT_ca-bundle=y
CONFIG_DEFAULT_ca-certificates=y
CONFIG_DEFAULT_cfdisk=y
CONFIG_DEFAULT_coremark=y
CONFIG_DEFAULT_curl=y
CONFIG_DEFAULT_ddns-scripts_aliyun=y
CONFIG_DEFAULT_ddns-scripts_dnspod=y
CONFIG_DEFAULT_default-settings=y
CONFIG_DEFAULT_dnsmasq-full=y
CONFIG_DEFAULT_dropbear=y
CONFIG_DEFAULT_e2fsprogs=y
CONFIG_DEFAULT_firewall=y
CONFIG_DEFAULT_fstools=y
CONFIG_DEFAULT_intel-microcode=y
CONFIG_DEFAULT_ip-full=y
CONFIG_DEFAULT_ipset=y
CONFIG_DEFAULT_iptables=y
CONFIG_DEFAULT_iptables-mod-extra=y
CONFIG_DEFAULT_iptables-mod-tproxy=y
CONFIG_DEFAULT_kmod-8139cp=y
CONFIG_DEFAULT_kmod-8139too=y
CONFIG_DEFAULT_kmod-amazon-ena=y
CONFIG_DEFAULT_kmod-amd-xgbe=y
CONFIG_DEFAULT_kmod-atlantic=y
CONFIG_DEFAULT_kmod-bnx2=y
CONFIG_DEFAULT_kmod-bnx2x=y
CONFIG_DEFAULT_kmod-button-hotplug=y
CONFIG_DEFAULT_kmod-drm-amdgpu=y
CONFIG_DEFAULT_kmod-drm-i915=y
CONFIG_DEFAULT_kmod-e1000=y
CONFIG_DEFAULT_kmod-e1000e=y
CONFIG_DEFAULT_kmod-forcedeth=y
CONFIG_DEFAULT_kmod-fs-f2fs=y
CONFIG_DEFAULT_kmod-fs-vfat=y
CONFIG_DEFAULT_kmod-i40e=y
CONFIG_DEFAULT_kmod-i40evf=y
CONFIG_DEFAULT_kmod-iavf=y
CONFIG_DEFAULT_kmod-igb=y
CONFIG_DEFAULT_kmod-igbvf=y
CONFIG_DEFAULT_kmod-igc=y
CONFIG_DEFAULT_kmod-ipt-raw=y
CONFIG_DEFAULT_kmod-ixgbe=y
CONFIG_DEFAULT_kmod-mlx4-core=y
[... 5279 lines cut ...]
# CONFIG_PACKAGE_crowdsec-firewall-bouncer is not set
# CONFIG_PACKAGE_cshark is not set
# CONFIG_PACKAGE_daemonlogger is not set
# CONFIG_PACKAGE_darkstat is not set
# CONFIG_PACKAGE_dawn is not set
# CONFIG_PACKAGE_dhcpcd is not set
# CONFIG_PACKAGE_dmapd is not set
# CONFIG_PACKAGE_dnscrypt-proxy2 is not set
# CONFIG_PACKAGE_dnsforwarder is not set
# CONFIG_PACKAGE_dnstap is not set
# CONFIG_PACKAGE_dnstop is not set
# CONFIG_PACKAGE_dpdk is not set
# CONFIG_PACKAGE_ds-lite is not set
# CONFIG_PACKAGE_dsmboot is not set
# CONFIG_PACKAGE_esniper is not set
CONFIG_PACKAGE_etherwake=y
# CONFIG_PACKAGE_etherwake-nfqueue is not set
CONFIG_PACKAGE_ethtool=y
# CONFIG_PACKAGE_ethtool-full is not set
# CONFIG_PACKAGE_external-protocol is not set
# CONFIG_PACKAGE_fail2ban is not set
# CONFIG_PACKAGE_fakeidentd is not set
# CONFIG_PACKAGE_fakepop is not set
# CONFIG_PACKAGE_family-dns is not set
# CONFIG_PAC

aiamadeus, 2024-11-24:
更新驱动之后呢

loydcheung, 2024-11-24:
> 更新驱动之后呢

感谢大佬持续关注。
更新驱动后能感觉坚持时间长一点，但是持续不过一两分钟后还是会整个系统卡死。
我现在怀疑是cpu的问题，照理说J4125不是很强但也不至于这个USB3.0的卡也hold不住吧。

zgc, 2025-04-21:
同样的问题，但是我的时长不定，最快的时候5钟就死机，最长一次跑了四个小时，cpu i3 1005G1，同样的驱动，现在准备换官方驱动试试

zgc, 2025-04-24:
[image]
更换驱动之后，已经稳定快2天了

coolsnowwolf, 2026-09-05:
讨论已验证：更换为官方 r8152 驱动后已有用户稳定运行近两天；原问题涉及 vendor 驱动/USB 转接硬件组合，非当前仓库无法定位的必现错误。当前也保留了标准 `kmod-usb-net-rtl8152` 选项，现关闭。

closed 2026-09-05 by coolsnowwolf