coolsnowwolf/lede #12522: TTYD内网地址打不开，通过域名能正常打开，是什么原因？

drw2005, 2024-10-06:
### 详细叙述

用192.168.1.1内网地址打开如下图，提示：192.168.1.1 发送的响应无效。但是通过域名访问能正常打开，这是什么原因？
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

Sun Oct  6 16:45:29 2024 daemon.notice ttyd[5760]: [2024/10/06 16:45:29:3035] N: __lws_lc_tag:  ++ [wsisrv|0|adopted] (1)
Sun Oct  6 16:45:29 2024 daemon.notice ttyd[5760]: [2024/10/06 16:45:29:3049] N: __lws_lc_untag:  -- [wsisrv|0|adopted] (0) 1.317ms
Sun Oct  6 16:45:29 2024 daemon.notice ttyd[5760]: [2024/10/06 16:45:29:3091] N: __lws_lc_tag:  ++ [wsisrv|1|adopted] (1)
Sun Oct  6 16:45:29 2024 daemon.notice ttyd[5760]: [2024/10/06 16:45:29:3103] N: __lws_lc_untag:  -- [wsisrv|1|adopted] (0) 1.202ms
Sun Oct  6 16:48:12 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:12:6212] N: __lws_lc_tag:  ++ [wsisrv|2|adopted] (1)
Sun Oct  6 16:48:12 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:12:6224] N: __lws_lc_untag:  -- [wsisrv|2|adopted] (0) 1.208ms
Sun Oct  6 16:48:12 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:12:6265] N: __lws_lc_tag:  ++ [wsisrv|3|adopted] (1)
Sun Oct  6 16:48:12 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:12:6278] N: __lws_lc_untag:  -- [wsisrv|3|adopted] (0) 1.259ms
Sun Oct  6 16:48:15 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:15:4998] N: __lws_lc_tag:  ++ [wsisrv|4|adopted] (1)
Sun Oct  6 16:48:15 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:15:5012] N: __lws_lc_untag:  -- [wsisrv|4|adopted] (0) 1.307ms
Sun Oct  6 16:48:15 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:15:5051] N: __lws_lc_tag:  ++ [wsisrv|5|adopted] (1)
Sun Oct  6 16:48:15 2024 daemon.notice ttyd[5760]: [2024/10/06 16:48:15:5065] N: __lws_lc_untag:  -- [wsisrv|5|adopted] (0) 1.405ms
Sun Oct  6 16:51:54 2024 daemon.notice ttyd[5760]: [2024/10/06 16:51:54:7176] N: __lws_lc_tag:  ++ [wsisrv|6|adopted] (1)
Sun Oct  6 16:51:54 2024 daemon.notice ttyd[5760]: [2024/10/06 16:51:54:7188] N: __lws_lc_untag:  -- [wsisrv|6|adopted] (0) 1.186ms
Sun Oct  6 16:51:54 2024 daemon.notice ttyd[5760]: [2024/10/06 16:51:54:7223] N: __lws_lc_tag:  ++ [wsisrv|7|adopted] (1)
Sun Oct  6 16:51:54 2024 daemon.notice ttyd[5760]: [2024/10/06 16:51:54:7236] N: __lws_lc_untag:  -- [wsisrv|7|adopted] (0) 1.292ms

lanruihua026, 2024-10-08:
重装ttyd软件包，尽量使用mobaxterm之类的ssh连接工具进入后台调试

xingshu1990, 2024-10-08:
浏览器输入:网址:7681
比如：
[image]

如果能显示TTYD的界面，说明没问题，
如果显示连接不上：

编辑/etc/init.d/ttyd 这个文件
给这行${interface:+-i $interface} 加个注释 或者删除。
然后/etc/init.d/ttyd restart 重启TTYD

coolsnowwolf, 2024-10-09:
其实是浏览器机制的问题。。。重启下路由器就行

drw2005, 2024-10-09:
> 其实是浏览器机制的问题。。。重启下路由器就行

大佬亲自回复，受宠若惊。
不过，我重启了也还是这种情况，之前都正常，就是最近出现的。

closed 2024-10-09 by drw2005