coolsnowwolf/lede #12523: 怎么移除firewall啊大佬

haoxueqing259, 2024-10-07:
### 详细叙述

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-ipt-nat for firewall
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	kmod-ipt-nat
 * 	iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for: 	firewall4 * 
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/hao/lede'
make[1]: *** [package/Makefile:111: /home/hao/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/hao/lede'
make: *** [/home/hao/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

小米路由器3

### 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-ipt-nat for firewall
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	kmod-ipt-nat
 * 	iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for: 	firewall4 * 
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/hao/lede'
make[1]: *** [package/Makefile:111: /home/hao/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/hao/lede'
make: *** [/home/hao/lede/include/toplevel.mk:231：world] 错误 2

lanruihua026, 2024-10-08:
因为不同的软件版本兼容性不同，在线安装90%都是失败的，想要装插件就编译的时候直接卸载掉，卸载插件则ssh下命令行执行opkg remove指令。

lanruihua026, 2024-10-08:
防火墙可以关闭，不一定要删除

xingshu1990, 2024-10-08:
遇事不决，重新编译。
如果是编译，建议第一次编译的时候，什么都不新增勾选，也不增加第三方固件地址，
就这样先简单的编译一遍，
等成功编译一次以后，再少量勾选几个插件进行编译，
后续可能疑似冲突的几款插件，勾选编译为ipk，
以IPK形式测试，查看是哪款插件需要安装firewall或者firewall4，与本身自带的firewall或者firewall4冲突了。
----
以下内容来自chatgpt3.5对楼主贴的日志进行的分析，得到的内容：

**错误分析**
缺少依赖：

您的系统无法找到 kmod-ipt-nat 和 iptables-mod-fullconenat 这两个软件包，这可能导致 firewall 软件包无法安装。
不兼容的架构：

提示中提到找到的 firewall 包与已配置的架构不兼容，说明您正在编译的环境和所需的软件包之间存在不匹配。
包冲突：

提示中还显示与 firewall4 软件包存在冲突，可能是因为您同时尝试安装不兼容的防火墙版本。

**解决步骤**
以下是一些可能的解决方案：

1. 检查配置
确保您的 OpenWrt 构建配置文件（.config）中选择了正确的架构和目标平台。您可以使用以下命令来查看和修改配置：

make menuconfig
在 Target System 和 Subtarget 选项中，确保选择了与您的硬件平台相匹配的选项。

2. 确保依赖包可用
确认 kmod-ipt-nat 和 iptables-mod-fullconenat 包在您选择的构建配置中是可用的。可以在 make menuconfig 中查找这些模块，并确保它们被选中。

3. 清理和重新编译
有时候构建过程中的残留文件会导致问题。可以尝试清理构建并重新编译：

make clean
make

4. 检查和更新源
确保您的代码库是最新的，并且所有包都来自于可用的源。您可以通过运行以下命令来更新：

git pull
然后再执行 make 进行编译。

5. 手动安装缺失的依赖
如果在编译过程中确实缺少某些依赖，您可以尝试手动下载并安装这些包，确保它们的版本与您的目标平台兼容。

haoxueqing259, 2024-10-08:
> 因为不同的软件版本兼容性不同，在线安装90%都是失败的，想要装插件就编译的时候直接卸载掉，卸载插件则ssh下命令行执行opkg remove指令。

好的，谢谢您的回复，后面我没有选firewall4编译成功了，但是不知道是不是小米路由器3的架构对我选的东西不支持，刷进去不能正常开机，无限重启，最后还是用上了其他人编译好的x-wrt固件，我主要就是想编译一个可以破解校园网多设备限制的系统，可能是我看的教程的问题，我没有找到适合我的路由器的编译教程，所以使用了x86架构路由器的编译教程，所以编译出了问题。

haoxueqing259, 2024-10-08:
> 遇事不决，重新编译。 如果是编译，建议第一次编译的时候，什么都不新增勾选，也不增加第三方固件地址， 就这样先简单的编译一遍， 等成功编译一次以后，再少量勾选几个插件进行编译， 后续可能疑似冲突的几款插件，勾选编译为ipk， 以IPK形式测试，查看是哪款插件需要安装firewall或者firewall4，与本身自带的firewall或者firewall4冲突了。 [...]

好的，谢谢您的回复，我应该就是编译了不兼容我路由器架构的功能，所以出现了问题，后面虽然编译成功了，刷到路由器之后。也是无法正常开机。

haoxueqing259, 2024-10-08:
感谢各位大佬的回复，谢谢，我找到了小米路由器3新的编译教程，暂时没有问题了。
https://anclark.github.io/2023/05/12/OpenWRT/OpenWRT_MiWifi3_PB-Boot/#:~:text=%E5%A2%9E%E5%8A%A0PB-Boot%E6%94%AF

closed 2024-10-08 by haoxueqing259