coolsnowwolf/lede #12528: 一直出现这个错误是咋回事？

drw2005, 2024-10-10:
### 详细叙述

Lede18.06 内核版本 6.6.54

昨天开始出现大量这个错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

Thu Oct 10 09:52:47 2024 daemon.err odhcp6c[10023]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Oct 10 09:54:45 2024 daemon.err odhcp6c[10023]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Oct 10 09:56:58 2024 daemon.err odhcp6c[10023]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Oct 10 09:59:06 2024 daemon.err odhcp6c[10023]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Oct 10 10:01:14 2024 daemon.err odhcp6c[10023]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)

yanxin152133, 2024-10-12:
Failed to send SOLICIT message to ff02::1:2 (Permission denied)
复制这句话直接google，看出来的结果哪条适合你
[image]

drw2005, 2024-10-12:
> Failed to send SOLICIT message to ff02::1:2 (Permission denied) 复制这句话直接google，看出来的结果哪条适合你 [...]

谢谢啦

closed 2024-10-12 by drw2005