coolsnowwolf/lede #12529: 自编译的树莓派3B+ 不能保存 挂载点异常

xingshu1990, 2024-10-11:
### 详细叙述

## 编译环境 及 使用设备环境
编译环境为vbox 安装的是ubuntu 22.04
使用环境为树莓派3B+ 通常编译的是32位

## 编译细节
首先，git clone，死活下载不了，遂手动下载zip，解压，切换到目录下：
./scripts/feeds update -a
./scripts/feeds install -a

第一次编译的时候，设置了这些：
[image]
[image]

以及> Utilities > Editors 勾选nano-full，
保存，
新建dl文件夹，解压自用的dl包到dl文件夹内，
执行make download -j8 V=s
没问题后，
再执行make V=s -j1。

出现过一次，丢给chatgpt3.5翻译，遂猜测可能是Target Images 中的Kernel partition size (in MiB) 和Root filesystem partition size (in MiB)  问题，重新进Target Images 修改，保存后重新make download -j8 V=s 
再输入make -j$(nproc) || make -j1 || make -j1 V=s 
[image]

有一个报错信息，chatgpt3.5翻译后，说是：
【报错来源：这个警告来源于 OpenWrt/LEDE 的包管理系统，在构建和打包时遇到与 usign 相关的 SHA-512 校验问题。】，
这点小问题先不纠结.
[image]

直接格式化、重新烧录TF卡，插机输入192.168.1.1，能正常进入openwrt，能将IP地址修改为新地址。
但是其他部分操作，修改后能保存，但是不能应用。

我个人猜测 是编译过程中勾选了sysupdate导致的问题，但是多数的编译教程里没提到这个问题，而且我有几次编译后烧录的固件也是没出现过问题，所以不能针对性的排查问题。

有问题的固件包img压缩包地址：
https://github.com/xingshu1990/BCM2710_dl/releases/tag/openwrt

----

以下文件日志较长,，
是运行日志，非编译日志：

```

Fri Oct 11 13:15:13 2024 kern.info kernel: [    0.323196] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.053353] Core Release: 2.80a
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.055435] Setting default values for core params
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.057507] Finished setting default values for core params
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.259773] Using Buffer DMA mode
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.261801] Periodic Transfer Interrupt Enhancement - disabled
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.263877] Multiprocessor Interrupt Enhancement - disabled
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.265938] OTG VER PARAM: 0, OTG VER FLAG: 0
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.267959] Dedicated Tx FIFOs mode
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.270458]
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.270464] WARN::dwc_otg_hcd_init:1072: FIQ DMA bounce buffers: virt = f0d0c000 dma = 0x00000000f7514000 len=9024
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.276195] FIQ FSM acceleration enabled for :
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.276195] Non-periodic Split Transactions
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.276195] Periodic Split Transactions
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.276195] High-Speed Isochronous Endpoints
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.276195] Interrupt/Control Split Transaction hack enabled
Fri Oct 11 13:15:13 2024 kern.debug kernel: [    1.284817] dwc_otg: Microframe scheduler enabled
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.284935]
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.284940] WARN::hcd_init_fiq:457: FIQ on core 1
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.288213]
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.288217] WARN::hcd_init_fiq:458: FIQ ASM at c0712718 length 36
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.291564]
Fri Oct 11 13:15:13 2024 kern.warn kernel: [    1.291568] WARN::hcd_init_fiq:496: MPHI regs_base at f0810000
Fri Oct 11 13:15:13 2024 kern.info kernel: [    1.294947] dwc_otg 3f980000.usb: DWC OTG Controller
[... 488 lines cut ...]
Fri Oct 11 13:21:55 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:55:7427] N: __lws_lc_tag:  ++ [wsisrv|0|adopted] (1)
Fri Oct 11 13:21:55 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:55:7432] N: HTTP / - 192.168.1.215
Fri Oct 11 13:21:55 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:55:8929] N: HTTP /token - 192.168.1.215
Fri Oct 11 13:21:55 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:55:9641] N: __lws_lc_tag:  ++ [wsisrv|1|adopted] (2)
Fri Oct 11 13:21:55 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:55:9648] N: WS   /ws - 192.168.1.215, clients: 1
Fri Oct 11 13:21:56 2024 daemon.notice ttyd[3521]: [2024/10/11 13:21:56:0015] N: started process, pid: 5095
Fri Oct 11 13:22:00 2024 daemon.notice ttyd[3521]: [2024/10/11 13:22:00:8988] N: __lws_lc_untag:  -- [wsisrv|0|adopted] (1) 5.156s
Fri Oct 11 13:22:56 2024 daemon.err ttyd[3521]: [2024/10/11 13:22:56:0057] E: read_cb: EIO (i/o error)
Fri Oct 11 13:22:56 2024 daemon.notice ttyd[3521]: [2024/10/11 13:22:56:0061] N: process exited with code 0, pid: 5095
Fri Oct 11 13:22:56 2024 daemon.notice ttyd[3521]: [2024/10/11 13:22:56:0065] N: WS closed from 192.168.1.215, clients: 0
Fri Oct 11 13:22:56 2024 daemon.notice ttyd[3521]: [2024/10/11 13:22:56:0068] N: __lws_lc_untag:  -- [wsisrv|1|adopted] (0) 1.000min
Fri Oct 11 13:26:20 2024 kern.info kernel: [  659.033872] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: disabled.
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Raspberry Pi 3 Model B Plus Rev 1 

### 详细日志

可能不重要

xingshu1990, 2024-10-11:
可能对于某些高手来说，我这个是常识性问题，问题比较低端。这个烦请见谅。

CYaNu, 2024-10-11:
1 /dev/mmcblk0 有磁盘错误 这个常见 在终端 umount -a 然后分别e2fsck /dev/mmcblk0p1和e2fsck /dev/mmcblk0p2 提示修复按y 然后重启
2 保存问题 在启动项里 最下面 exit 0前加一行mount -o rw,remount  /dev/mmcblk0p2 保存后重启
要学会都log 都在里面

xingshu1990, 2024-10-12:
已经把很多的log内容丢给chatgpt3.5去分析了，且也有过几个修复的操作（比如按照chatgpt3.5提供的方法，取消挂载，然后修复，重新挂载），但是这个本身应该是编译前就要做的操作，不应该是编译后全新的系统，然后花N多时间找到了多个BUG，你说 是吧？而且我这样修复了以后，还是一样不能保存并应用。

CYaNu, 2024-10-12:
说实话 你不适合玩这个

coolsnowwolf, 2024-10-12:
什么固件需要 3GB 的根目录。。。？

xingshu1990, 2024-10-12:
> 什么固件需要 3GB 的根目录。。。？

你就别纠结这个事情了[/捂脸]，原先最开始看视频折腾编译的时候，视频作者为了方便，直接在Root filesystem partition size (in MiB) 后面加了个0，另外原先我折腾newifi y1挂载的U盘是8G以上的盘什么的，图方便搞的。