coolsnowwolf/lede #12534: 小米路由器3g采用闭源驱动编译后wifi没有网速

gforce1206, 2024-10-13:
### 详细叙述

默认内核5.4，make menucofing：
LuCI—>Applications—> luci-app-mtwifi
Network—>wireless—>mt_wifi—>MT_WIFI Configuration—>MTK_CHIP_MT7603E_MT7612E
Kernel modules—>Wireless Drivers—>kmod-mt7603e、kmod-mt76x2e
取消：Network —>wirelessAPD—>wpad-openssl
编译出来后插在光猫上做二级路由器测试，2.4g及5g WiFi均无速度，而测试了开源驱动编译，WiFi正常

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xiaomi-r3g

### 详细日志

无

gforce1206, 2024-10-26:
大概找到问题了，参照这个帖子[https://www.right.com.cn/forum/thread-8355729-1-1.html](url)
在`target\linux\ramips\base-files\etc\board.d\02_network`下
`xiaomi,redmi-router-ac2100)
      ucidef_add_switch "switch0" \
            "2:lan:1" "3:lan:2" "4:lan:3" "0:wan" "6u@eth0" "5u@eth1"
      ;;`
要有带有switch0的机器才正常，反正编译出来的固件WiFi没有网速，小白，看着像是交换机的问题

coolsnowwolf, 2026-09-05:
已定位并修正配置保护。Mi Router 3G 设备配置默认已选择开源 kmod-mt7603 和 kmod-mt76x2；issue 中又同时选择了闭源 kmod-mt7603e 和 kmod-mt76x2e，会让两套驱动竞争同一组 PCI 无线设备（MT7603 模块名甚至同为 mt7603e），因此会出现无线可见但数据面异常。

- 现为两个闭源包加入了与对应开源驱动的显式冲突，阻止生成这种无效组合；
- ramips/mt7621 Linux 5.10.269 内核完整编译通过；
- 两个闭源包均编译通过：kmod-mt7603e（856402 bytes）、kmod-mt76x2e（599494 bytes）。
- 如确实要使用闭源驱动，需要从设备默认包中移除开源 kmod-mt7603/kmod-mt76x2，不能同时选中。

无线包修复：https://github.com/coolsnowwolf/lede/commit/cf84d2c229792b91012e11998355a117ee5cd376
同时修复了测试时发现的 5.10/6.18 文件覆盖回归：https://github.com/coolsnowwolf/lede/commit/2c34fbaa2709604d6ce9391fb4f4cd9ead74b807。评论中添加 switch0 的方案属于旧 swconfig；当前该机型使用 DSA，不应恢复 switch0。

closed 2026-09-05 by coolsnowwolf