coolsnowwolf/lede #12535: Manjaro (Arch Linux) 源码编译报错 & ERROR: toolchain/gcc/initial failed to build.

WenqiOfficial, 2024-10-13:
### 详细叙述

使用 Manjaro 编译LEDE源码报错，看log应该是准备工具链时就失败了

编译环境系统内核：6.10.13-3-MANJARO
编译repo版本：2024.10.13 19:38

使用了 openwrt-devel 准备环境

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

所有可编译型号

### 详细日志

```
make[2]: Entering directory '/home/wenqi/openwrt/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/wenqi/openwrt/lede/scripts/config'
make[1]: Entering directory '/home/wenqi/openwrt/lede'
make[2]: Entering directory '/home/wenqi/openwrt/lede'
+ mkdir -p /home/wenqi/openwrt/lede/staging_dir/target-x86_64_musl
+ cd /home/wenqi/openwrt/lede/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/wenqi/openwrt/lede/build_dir/target-x86_64_musl/stamp
touch /home/wenqi/openwrt/lede/staging_dir/target-x86_64_musl/.prepared
make[3]: Entering directory '/home/wenqi/openwrt/lede/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/wenqi/openwrt/lede/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.00
make[3]: Entering directory '/home/wenqi/openwrt/lede/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/wenqi/openwrt/lede/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/wenqi/openwrt/lede/toolchain/gcc/initial'
export SHELL="bash"; make CFLAGS="-O2 -I/home/wenqi/openwrt/lede/staging_dir/host/include " CFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CXXFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" GOCFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro"  -C /home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial all-build-libiberty all-gcc all-target-libgcc
make[4]: Entering directory '/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[5]: Entering directory '/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
if [ x"" != x ]; then \
  x86_64-pc-linux-gnu-gcc -c -DHAVE_CONFIG_H   -I. -I/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection  /home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o pic/simple-object-mach-o.o; \
else true; fi
if [ x"" != x ]; then \
  x86_64-pc-linux-gnu-gcc -c -DHAVE_CONFIG_H   -I. -I/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection   /home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o noasan/simple-object-mach-o.o; \
else true; fi
x86_64-pc-linux-gnu-gcc -c -DHAVE_CONFIG_H   -I. -I/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection /home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o simple-object-mach-o.o
/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c: In function 'simple_object_mach_o_write_segment':
/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c:1231:17: error: passing argument 1 of 'set_32' from incompatible pointer type [-Wincompatible-pointer-types]
 1231 |         set_32 (&index[i], index[i]);
      |                 ^~~~~~~~~
      |                 |
      |                 unsigned int *
/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c:1231:17: note: expected 'unsigned char *' but argument is of type 'unsigned int *'
make[5]: *** [Makefile:1308: simple-object-mach-o.o] Error 1
make[5]: Leaving directory '/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[4]: *** [Makefile:2756: all-build-libiberty] Error 2
make[4]: Leaving directory '/home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/wenqi/openwrt/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/wenqi/openwrt/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.06#0.00#0.07
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/wenqi/openwrt/lede'
make[1]: *** [toolchain/Makefile:93: /home/wenqi/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/wenqi/openwrt/lede'
make: *** [/home/wenqi/openwrt/lede/include/toplevel.mk:231：world] 错误 2
```

aiamadeus, 2024-10-13:
docker 起个 debian 吧

WenqiOfficial, 2024-10-13:
sad....

确实可以这样暂时解决一下啦，谢谢w

但是还是希望能直接在arch系上完成编译...

WenqiOfficial, 2024-10-15:
感谢！

这个commit成功解决了arch系编译的问题！

closed 2024-10-15 by WenqiOfficial