coolsnowwolf/lede #12540: 请教大佬，这个叫什么电台的插件如何去掉？

drw2005, 2024-10-16:
### 详细叙述

编译完成发现有一个叫电台的插件
[image]
menuconfig找了一下，是不是这个啊？
[image]
但是箭头处好像不让去掉，下面的两个选项去掉后，编译出来仍然有这个插件。
[image]

小白请教，应该如何去掉它呢？
感谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

无

xingshu1990, 2024-10-16:
感觉你是用了其他第三方软件源了吧。
L大的固件，默认不会编译这个东西，最近几天摸鱼编译了3-4天的初始固件
编译出来的固件，没看到这个东西。

drw2005, 2024-10-16:
> 感觉你是用了其他第三方软件源了吧。 L大的固件，默认不会编译这个东西，最近几天摸鱼编译了3-4天的初始固件 编译出来的固件，没看到这个东西。

是的，用的这个：https://github.com/cachenow/openwrt-packages-storage
找不到和他关联的包，取消不了啊，想来看看有没大佬知道的。

xingshu1990, 2024-10-17:
建议先备份dl目录，然后将环境恢复成干净的初始状态，
```
make clean
清理bin目录下生成的烧录镜像文件

make dirclean
【make clean】+ 清理toolchain目录和目录中的（交叉）编译工具

make distclean
【make dirclean】+ 清除所有原目录后来产生的文件，包括下载的软件包，配置文件，feed内容等。

```
先编译干净的初始包，备份，
再编译一个未使用第三方软件包，但添加了自己需要的功能的固件，备份。
再按照https://www.youtube.com/watch?v=FKyqTtbU_LU 单独编译法，进行编译。
注：实际操作过程中，会出现找不到实际目录的情况，这个我没法给你具体命令，
我也是丢给chatgpt多次尝试才编译了ipk，不过这个IPK也还得自己再编译其他关联包。

zheshifandian, 2024-10-17:
nmdvm-luci下边两个取消掉，然后↑↓选到nmdvm-luci，菜单下边有个help，回车，看看Selected by哪个包，取消掉就行了

drw2005, 2024-10-17:
> 建议先备份dl目录（后面进窗口勾选保存.config后，可把这个dl文件夹复制到L大源码根目录下），然后将环境恢复成干净的初始状态，
> 
> ```
> make clean
> 清理bin目录下生成的烧录镜像文件
> 
> make dirclean
> 【make clean】+ 清理toolchain目录和目录中的（交叉）编译工具
> 
> make distclean
> 【make dirclean】+ 清除所有原目录后来产生的文件，包括下载的软件包，配置文件，feed内容等。
> ```
> 
> 然后按照L大教程里的流程 编译干净的初始包，备份，以便刷乱了，自用。 再编译一个未使用第三方软件包，但添加了自己需要的功能的固件，备份，以便刷乱了，自用。
> 
> 再按照https://www.youtube.com/watch?v=FKyqTtbU_LU 单独编译法，进行编译。 注：实际操作过程中，会出现找不到实际目录的情况，这个我没法给你具体命令， 我也是丢给chatgpt多次尝试才编译了ipk，不过这个IPK也还得自己再编译其他关联包。

谢谢指点，我试试。

drw2005, 2024-10-17:
> nmdvm-luci下边两个取消掉，然后↑↓选到nmdvm-luci，菜单下边有个help，回车，看看Selected by哪个包，取消掉就行了

是这个，但是找不到在哪里
[image]

zheshifandian, 2024-10-17:
> > nmdvm-luci下边两个取消掉，然后↑↓选到nmdvm-luci，菜单下边有个help，回车，看看Selected by哪个包，取消掉就行了
> 
> 是这个，但是找不到在哪里 ![QQ截图20241017153218](https://private-user-images.githubusercontent.com/33473981/377358458-e00eeadd-4279-4ce2-8c38-756466ff88e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkxNTk2MTEsIm5iZiI6MTcyOTE1OTMxMSwicGF0aCI6Ii8zMzQ3Mzk4MS8zNzczNTg0NTgtZTAwZWVhZGQtNDI3OS00Y2UyLThjMzgtNzU2NDY2ZmY4OGU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE3VDEwMDE1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzZDA5NzczNDI4MWMyZjQ3ZDgwMTAxNWFkZjJmNGIzNWU5NzE1MWE5ZjI3M2MwOTBkZGU5NjgyODZiYzVmMTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JipqrJHB8BIMKzW3PxG8HhbuBpV0EKkMoZY0AONUFcQ)

make menuconfig后，按 / ，输入mmdvm，回车就知道在哪了

selboo, 2024-10-18:
或者直接删除。然后刷新就没了。

```
opkg remove libmmdvm mmdvm-host mmdvm-luci mmdvm-luci-i18n-zh-cn

opkg remove libmmdvm mmdvm-host mmdvm-luci mmdvm-luci-i18n-zh-cn

opkg remove libmmdvm mmdvm-host mmdvm-luci mmdvm-luci-i18n-zh-cn

```

多执行几遍。直到提示 `No packages removed. `

closed 2024-10-24 by drw2005

Mike-qian, 2024-11-21:
执行完./scripts/feeds install -a后执行下面这段
./scripts/feeds uninstall {mmdvm,mmdvm-host,p25-clients,ysf-clients,nxdn-clients,mmdvm-luci,dapnet-gateway}