coolsnowwolf/lede #12542: 请问这个错误怎么解决

GreatMichaelLee, 2024-10-16:
### 详细叙述

Collected errors:
 * check_data_file_clashes: Package luci-lib-ipkg wants to install file /home/builder/lede_x86/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/model/ipkg.lua
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package luci-app-store.
 * check_data_file_clashes: Package luci-lib-ipkg wants to install file /home/builder/lede_x86/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/model/ipkg.lua
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package luci-lib-ipkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:111: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

见日志

xingshu1990, 2024-10-17:
chatgpt解释：你遇到的问题是由于文件冲突引起的。luci-lib-ipkg包试图安装一个文件 (/usr/lib/lua/luci/model/ipkg.lua)，但该文件已经由luci-base包提供。这个冲突阻止了luci-lib-ipkg和luci-app-store的安装，导致构建过程失败。

---
如果不在乎重新花时间编译
建议备份dl文件夹到其他地方（重新编译时，make download的时候，可以重新利用起来）,
然后试着make dirclean或make distclean：

make dirclean
【make clean】+ 清理toolchain目录和目录中的（交叉）编译工具

make distclean
【make dirclean】+ 清除所有原目录后来产生的文件，包括下载的软件包，配置文件，feed内容等。

然后先试着初期编译一个干净的固件，然后把这个固件文件先移动到其他地方，
再新编译新增较少功能的固件，
另，反正是新编译的状态下，可以先试试在干净固件的基础上，找个你想要的ipk文件进行安装，
测试是否是ipk冲突。

GreatMichaelLee, 2024-10-18:
> chatgpt解释：你遇到的问题是由于文件冲突引起的。luci-lib-ipkg包试图安装一个文件 (/usr/lib/lua/luci/model/ipkg.lua)，但该文件已经由luci-base包提供。这个冲突阻止了luci-lib-ipkg和luci-app-store的安装，导致构建过程失败。
> 
> 如果不在乎重新花时间编译 建议备份dl文件夹到其他地方（重新编译时，make download的时候，可以重新利用起来）, 然后试着make dirclean或make distclean：
> 
> make dirclean 【make clean】+ 清理toolchain目录和目录中的（交叉）编译工具
> 
> make distclean 【make dirclean】+ 清除所有原目录后来产生的文件，包括下载的软件包，配置文件，feed内容等。
> 
> 然后先试着初期编译一个干净的固件，然后把这个固件文件先移动到其他地方， 再新编译新增较少功能的固件， 另，反正是新编译的状态下，可以先试试在干净固件的基础上，找个你想要的ipk文件进行安装， 测试是否是哪个ipk，或者依赖库（依赖软件）冲突。

不管用，没解决我的问题，我rm -rf ./tmp && make clean后，导入配置问题依旧

xingshu1990, 2024-10-18:
> 不管用，没解决我的问题，我rm -rf ./tmp && make clean后，导入配置问题依旧

还是等大佬吧。

我个人是觉得原先勾选保存.config，得删.config，先试着编译默认配置的固件试试 编译生成的编译环境是否已经有问题。

GreatMichaelLee, 2024-10-19:
> 还是等大佬吧。
>
> 我个人是觉得原先勾选保存.config，得删.config，先试着编译默认配置的固件试试 编译生成的编译环境是否已经有问题。

我同样的.config+feeds.conf.default在另一台机器上完全没问题啊，关键是

yujieli001, 2024-11-20:
一模一样的问题，新下载编译的，以前编译都很正常，用的是Ubuntu22.04 lts。
make[3]: Leaving directory '/sites/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#78.67#18.42#98.43
make[2]: Leaving directory '/sites/lede'
make[1]: *** [package/Makefile:110: /sites/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/sites/lede'
make: *** [/sites/lede/include/toplevel.mk:231: world] Error 2

closed 2024-11-23 by coolsnowwolf