coolsnowwolf/lede #12545: make download -j8后出现uugamebooster failed to build

hhhhhhheg, 2024-10-17:
### 详细叙述

make download -j8
……
ERROR: package/feeds/packages/uugamebooster failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ea6350v3

### 详细日志

无

binge8, 2024-10-17:
不放日志建议找个算命先生算算就解决了

bob-zebedy, 2024-10-17:
给我开个 ssh，我上去帮你看看

hhhhhhheg, 2024-10-18:
> 不放日志建议找个算命先生算算就解决了 不会英语也不会找个翻译工具看看？人家明确提示你了请用 -j1 V=s 或 V=sc 重新运行 make，以获得更高的详细级别，看看发生了什么。
第一次接触这个，这段代码怎么用啊，是加在make download 后面吗

hhhhhhheg, 2024-10-18:
> 不放日志建议找个算命先生算算就解决了 不会英语也不会找个翻译工具看看？ 人家明确提示你了Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on。

make[2]: Entering directory '/home/hhhhhhheg/Desktop/lede/feeds/packages/net/uugamebooster'
mkdir -p /home/hhhhhhheg/Desktop/lede/dl
SHELL= flock /home/hhhhhhheg/Desktop/lede/tmp/.uugamebooster-arm-v7.9.14.tar.gz.flock -c '  	/home/hhhhhhheg/Desktop/lede/scripts/download.pl "/home/hhhhhhheg/Desktop/lede/dl" "uugamebooster-arm-v7.9.14.tar.gz" "2e624deea128a298fce71f243d92092dad23451de7f547ad82008b9dd989bca5" "" "https://uu.gdl.netease.com/uuplugin/openwrt-arm/v7.9.14/uu.tar.gz?"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://uu.gdl.netease.com/uuplugin/openwrt-arm/v7.9.14/uu.tar.gz?/uugamebooster-arm-v7.9.14.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
curl: (22) The requested URL returned error: 403 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/uugamebooster-arm-v7.9.14.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/uugamebooster-arm-v7.9.14.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/uugamebooster-arm-v7.9.14.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[2]: *** [Makefile:86: /home/hhhhhhheg/Desktop/lede/dl/uugamebooster-arm-v7.9.14.tar.gz] Error 2
make[2]: Leaving directory '/home/hhhhhhheg/Desktop/lede/feeds/packages/net/uugamebooster'
time: package/feeds/packages/uugamebooster/download#0.23#0.06#18.95
    ERROR: package/feeds/packages/uugamebooster failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/uugamebooster/download] Error 1
make[1]: Leaving directory '/home/hhhhhhheg/Desktop/lede'
make[1]: Entering directory '/home/hhhhhhheg/Desktop/lede'
make[2]: Entering directory '/home/hhhhhhheg/Desktop/lede/target/linux'
make[3]: Entering directory '/home/hhhhhhheg/Desktop/lede/target/linux/ipq40xx'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/hhhhhhheg/Desktop/lede/target/linux/ipq40xx'
make[2]: Leaving directory '/home/hhhhhhheg/Desktop/lede/target/linux'
time: target/linux/download#0.07#0.01#0.08
make[1]: Leaving directory '/home/hhhhhhheg/Desktop/lede'

xingshu1990, 2024-10-19:
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
手动下载
1.换下科学上网的节点，重新编译
2.手动下载https://sources.cdn.openwrt.org/uugamebooster-arm-v7.9.14.tar.gz到dl目录，然后继续编译。

hhhhhhheg, 2024-10-19:
> ## curl: (22) The requested URL returned error: 404 Not Found
> Download failed.
> 1.换下科学上网的节点，重新编译 2.手动下载[uugamebooster-arm-v7.9.14.tar.gz](https://uu.gdl.netease.com/uuplugin/openwrt-arm/v7.9.14/uu.tar.gz?/uugamebooster-arm-v7.9.14.tar.gz) ，注意改名为uugamebooster-arm-v7.9.14.tar.gz 到/lede/dl目录，然后重新编译。

今天搞的时候又好了…昨天第一个链接还是报403，挺无语的，还是希望能有稳定的链接能下载吧😂

closed 2024-10-19 by hhhhhhheg