coolsnowwolf/lede #12547: 更新后进不去后台

1105384898, 2024-10-18:
### 详细叙述

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/argone/header'.
A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '__entries' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:68: in function 'node'
[string "/usr/lib/lua/luci/view/themes/argone/header..."]:20: in main chunk

In error(), file [C]
called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `            die(ex);`
  Near here --------^

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

kxdn3, 2024-10-18:
一样，悲剧

kxdn3, 2024-10-18:
怎么救一救…不想资料没了

xingshu1990, 2024-10-18:
## 我自己可能的办法，尝试重装luci

## 以下来自chatgpt，可以是一种尝试：

你遇到的错误与 OpenWrt 的 LuCI Web 界面有关，具体是关于 `themes/argone/header` 模板文件在渲染过程中发生了异常。这种情况通常是由于某个模板文件中的变量或依赖项没有正确初始化，或者 Lua 代码中存在语法或逻辑问题导致的。

错误的关键提示如下：
- **`attempt to index global '__entries' (a nil value)`**：这是在访问一个名为 `__entries` 的全局变量时，系统发现它是 `nil`（空值），因此无法继续操作。
- **涉及的文件**：
  - `/usr/lib/lua/luci/dispatcher.lua`
  - `/usr/lib/lua/luci/template.lua`
  - `/usr/lib/lua/luci/ucodebridge.lua`

### 可能的解决方法：

1. **检查主题文件**：
   - 错误提示指出了 `themes/argone/header` 模板文件。你可以检查这个文件是否存在问题，尤其是看它是否正确加载了所有需要的 Lua 变量或函数。如果你对 Lua 编程不熟悉，可以尝试切换到 OpenWrt 的默认主题，看是否能够避免此错误。

   - 尝试切换到默认主题，可以通过 SSH 连接执行以下命令：
     ```bash
     uci set luci.main.mediaurlbase='/luci-static/bootstrap'
     uci commit luci
     /etc/init.d/uhttpd restart
     ```
     这将把主题切换到默认的 `bootstrap` 主题，看看问题是否解决。

2. **更新 LuCI 组件**：
   - 如果 `luci` 包有损坏或丢失的依赖，可能需要重新安装或更新相关组件。你可以尝试运行以下命令来更新 `luci` 和相关依赖：
     ```bash
     opkg update
     opkg install luci-theme-bootstrap --force-reinstall
     opkg install luci --force-reinstall
     ```

3. **检查 Lua 文件**：
   - 错误也指出了 `/usr/lib/lua/luci/dispatcher.lua:68` 中的变量未正确初始化。你可以查看该文件中第 68 行附近的代码，看看是否有异常。如果可能，这部分代码可能需要修复。

4. **重启 LuCI 或设备**：
   - 有时，仅仅是因为某些缓存或配置未正确加载，导致了这个错误。你可以尝试重启 LuCI 服务或设备：
     ```bash
     /etc/init.d/uhttpd restart
     ```
   - 如果重启服务无效，也可以考虑重启整个设备，看看是否解决问题。

### 如果问题依然存在：
- 如果以上步骤未能解决问题，试着在原先的固件编译基础上，先不勾选【argone】这个模板进行编译。

1105384898, 2024-10-18:
> 怎么救一救…不想资料没了
我等更新哈哈，然后编译完把包用ftp传软路由上，用命令行sysupgrade -v 包名去升级，我也不懂这个技术，不知道咋解决，只能等修复

1105384898, 2024-10-18:
> ## 我自己可能的办法，尝试重装luci
> ## 以下来自chatgpt，可以是一种尝试：
> 你遇到的错误与 OpenWrt 的 LuCI Web 界面有关，具体是关于 `themes/argone/header` 模板文件在渲染过程中发生了异常。这种情况通常是由于某个模板文件中的变量或依赖项没有正确初始化，或者 Lua 代码中存在语法或逻辑问题导致的。
> 
> 错误的关键提示如下：
> 
> * **`attempt to index global '__entries' (a nil value)`**：这是在访问一个名为 `__entries` 的全局变量时，系统发现它是 `nil`（空值），因此无法继续操作。
> * **涉及的文件**：
>   
>   * `/usr/lib/lua/luci/dispatcher.lua`
>   * `/usr/lib/lua/luci/template.lua`
>   * `/usr/lib/lua/luci/ucodebridge.lua`
> 
> ### 可能的解决方法：
> 1. **检查主题文件**：
>    
>    * 错误提示指出了 `themes/argone/header` 模板文件。你可以检查这个文件是否存在问题，尤其是看它是否正确加载了所有需要的 Lua 变量或函数。如果你对 Lua 编程不熟悉，可以尝试切换到 OpenWrt 的默认主题，看是否能够避免此错误。
>    * 尝试切换到默认主题，可以通过 SSH 连接执行以下命令：
>      ```shell
>      uci set luci.main.mediaurlbase='/luci-static/bootstrap'
>      uci commit luci
>      /etc/init.d/uhttpd restart
>      ```
>      
>      
>          
>            
>          
>      
>            
>          
>      
>          
>        
>      这将把主题切换到默认的 `bootstrap` 主题，看看问题是否解决。
> 2. **更新 LuCI 组件**：
>    
>    * 如果 `luci` 包有损坏或丢失的依赖，可能需要重新安装或更新相关组件。你可以尝试运行以下命令来更新 `luci` 和相关依赖：
>      ```shell
>      opkg update
>      opkg install luci-theme-bootstrap --force-reinstall
>      opkg install luci --force-reinstall
>      ```
> 3. **检查 Lua 文件**：
>    
>    * 错误也指出了 `/usr/lib/lua/luci/dispatcher.lua:68` 中的变量未正确初始化。你可以查看该文件中第 68 行附近的代码，看看是否有异常。如果可能，这部分代码可能需要修复。
> 4. **重启 LuCI 或设备**：
>    
>    * 有时，仅仅是因为某些缓存或配置未正确加载，导致了这个错误。你可以尝试重启 LuCI 服务或设备：
>      ```shell
>      /etc/init.d/uhttpd restart
>      ```
>    * 如果重启服务无效，也可以考虑重启整个设备，看看是否解决问题。
> 
> ### 如果问题依然存在：
> * 如果以上步骤未能解决问题，试着在原先的固件编译基础上，先不勾选【argone】这个模板进行编译。

上面的试了不行，去掉模板下班编译看看行不行

lirg-111, 2024-10-18:
luci主题不要选argone

coolsnowwolf, 2024-10-18:
别用老版本 lua 主体在 js 版本上

1105384898, 2024-10-18:
> 别用老版本 lua 主体在 js 版本上

好的大佬

closed 2024-10-18 by 1105384898

kxdn3, 2024-10-18:
就是要换主题咯？

ihuaer, 2024-10-19:
知道了，看来要换js版本主题 了

1105384898, 2024-10-19:
> 知道了，看来要换js版本主题 了
对现在估计只有argon支持，我去掉其他主题，能正常进去了