coolsnowwolf/lede #12549: 最新的源码更新到24.10.24之后没有适配firewall4防火墙吗？

wixxm, 2024-10-18:
### 详细叙述

如题，更新了新的源码之后第一时间重新编译，但是发现没有对firewall4的支持，还是用不了很多新插件。。。是我设置的问题还是源码没有支持额
[image]
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

如题

hcym, 2024-10-18:
兵荒马乱的，就等等吧
不过我今天编译的正常使用，只有一个haproxy没有luci，基本没影响

wixxm, 2024-10-18:
> 兵荒马乱的，就等等吧 不过我今天编译的正常使用，只有一个haproxy没有luci，基本没影响

你更新之后编译出来的版本好是多少，我的怎么是24.9.18 不是 24.10.24

coolsnowwolf, 2024-10-18:
这不还没到 10/24 号嘛~~  目前以更成熟的 fw3 为主， fw4 需要自行选择

closed 2024-10-18 by coolsnowwolf