coolsnowwolf/lede #12550: upnp有规则但不通

harrison-guo-chn, 2024-10-18:
### 详细叙述

upnp在启用后有规则但是端口无法访问，具体设置如下

[image]
[image]

有公网ip，但是无法访问，尝试v1或者v2都是这样
[image]
[image]

测试过使用小米ax6000一切正常，无法找到解决方案
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-	OpenWrt R24.8.1 / LuCI Master (git-24.182.36151-66d29de)

### 详细日志

Fri Oct 18 23:36:38 2024 daemon.notice miniupnpd[3538]: shutting down MiniUPnPd
Fri Oct 18 23:36:38 2024 daemon.err uhttpd[2688]: Command failed: ubus call service delete { "name": "miniupnpd" } (Not found)
Fri Oct 18 23:36:43 2024 daemon.notice miniupnpd[19547]: HTTP listening on port 5000
Fri Oct 18 23:36:43 2024 daemon.notice miniupnpd[19547]: Listening for NAT-PMP/PCP traffic on port 5351

Travel2Here, 2024-10-21:
同。而且我用的还是老一点的版本R23.10.1
包括日志都一样

HollyMadCow, 2024-11-20:
这个应该用到转发吧

yanxin152133, 2024-11-21:
https://github.com/yanxin152133/lede-x86_x64/issues/1
试试我这样配置，里面有软件可以测试upnp
[image]

Mike-qian, 2024-11-23:
删掉自带的luci-app-upnp和miniupnpd下载个[X86内核新版](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这俩按上

Mike-qian, 2024-11-24:
> > 删掉自带的 luci-app-upnp 和 miniupnpd 下载个[X86 内核新版本](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这两者按上
> 
> 哪里下载啊，打不开，我就剩这个由问题了

https://dl.openwrt.ai/packages-24.10/packages/i386_pentium4/kiddin9/

Mike-qian, 2024-11-24:
> > 删掉自带的 luci-app-upnp 和 miniupnpd 下载个[X86 内核新版本](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这两者按上
> 
> 哪里下载啊，打不开，我就剩这个由问题了
你要是不是x86去https://dl.openwrt.ai/packages-24.10/packages/ 挑选自己的下载

Mike-qian, 2024-11-24:
> > > > 删掉自带的 luci-app-upnp 和 miniupnpd 下载个[X86 内核新版本](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这两者按上
> > > 
> > > 
> > > 哪里下载啊，打不开，我就剩这个由问题了
> > > 你要是不是x86去https://dl.openwrt.ai/packages-24.10/packages/ 挑选自己的下载
> 
> 不知道什么原因,我是x86平台的, 选了x86_64,下载安装提示 Unknown package

32位是i386

XinSSS, 2024-11-24:
> > > > > 删掉自带的 luci-app-upnp 和 miniupnpd 下载个[X86 内核新版本](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这两者按上
> > > > 
> > > > 
> > > > 哪里下载啊，打不开，我就剩这个由问题了
> > > > 你要是不是x86去https://dl.openwrt.ai/packages-24.10/packages/ 挑选自己的下载
> > 
> > 
> > 不知道什么原因,我是x86平台的, 选了x86_64,下载安装提示 Unknown package
> 
> 32位是i386

应该是64位的吧？我首页目标平台显示是x86/64

EdoQ, 2024-11-25:
我的是miniupnp正常，映射的端口也通，但是luci-app-upnp不显示已添加的转发

yanxin152133, 2024-11-25:
> 我的是miniupnp正常，映射的端口也通，但是luci-app-upnp不显示已添加的转发

是不是没被使用就不会显示

EdoQ, 2024-11-25:
> > 我的是miniupnp正常，映射的端口也通，但是luci-app-upnp不显示已添加的转发
> 
> 是不是没被使用就不会显示
不是的，我这边是通的，用最新源码编译的，我在配置文件里添加了option external_iface 'wan'，然后端口检测是通的，cat /var/upnp.leases也有规则，但是luci上就是不显示
[image]
[image]
[image]
[image]

Mike-qian, 2024-11-25:
> > > > > > 删掉自带的 luci-app-upnp 和 miniupnpd 下载个[X86 内核新版本](https://dl.openwrt.ai/24.10/packages/i386_pentium4/kiddin9)的这两者按上
> > > > > 
> > > > > 
> > > > > 哪里下载啊，打不开，我就剩这个由问题了
> > > > > 你要是不是x86去https://dl.openwrt.ai/packages-24.10/packages/ 挑选自己的下载
> > > 
> > > 
> > > 不知道什么原因,我是x86平台的, 选了x86_64,下载安装提示 Unknown package
> > 
> > 
> > 32位是i386
> 
> 应该是64位的吧？我首页目标平台显示是x86/64

下面这段加到软件包里手动升级看看
src/gz openwrt_kiddin9 https://dl.openwrt.ai/packages-24.10/packages/x86_64/kiddin9

XinSSS, 2024-11-26:
@Mike-qian 
必须把nft和ip6tables依赖安装全吗？奇怪，默认需要的话怎么会没有
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-nft-compat for xtables-nft
 * pkg_hash_fetch_best_installation_candidate: Packages for xtables-nft found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kmod-ip6tables for ip6tables-nft
 * pkg_hash_check_unresolved: cannot find dependency kmod-ip6tables for ip6tables-zz-legacy
 * pkg_hash_fetch_best_installation_candidate: Packages for ip6tables found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kmod-nf-conntrack-netlink for libnetfilter-conntrack3
 * pkg_hash_fetch_best_installation_candidate: Packages for libnetfilter-conntrack3 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for miniupnpd-iptables:
 * 	kmod-ip6tables
 * 	kmod-nf-conntrack-netlink
 * opkg_install_cmd: Cannot install package miniupnpd-iptables.
```
[image]

Mike-qian, 2024-11-26:
> ip6tables

ip6tables是ipv6才需要安装lede默认不按，带nft是4代防火墙的1代不用按，这些都可以忽视直接安装你miniupnpd-iptables安装不了就保留原来的miniupnpd看看

coolsnowwolf, 2026-09-05:
已修复。LEDE 的 miniupnpd-iptables 此前仍停留在 2.0.20170421，且旧 firewall3 脚本没有把 MINIUPNPD 链挂到外网 zone 的 input 链。现已同步上游实现：

- miniupnpd-iptables 升级到 2.3.9；
- 更新 init/hotplug/config，并加入 input、forward、prerouting、postrouting 的正确规则顺序；
- 移除仅适用于 2017 源码的旧补丁；
- x86_64 完整包编译通过：miniupnpd-iptables_2.3.9-3_x86_64.ipk（78589 bytes），打包脚本语法检查通过。
提交：https://github.com/coolsnowwolf/packages/commit/5c1409b475885a9a363318737bc6c2659bc419cb

closed 2026-09-05 by coolsnowwolf