coolsnowwolf/lede #12554: 新版luci，upnp插件ip不显示

1105384898, 2024-10-19:
### 详细叙述

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

1105384898, 2024-10-19:
[image]

DanielChen008, 2024-10-22:
请教：这是选择哪个主题的 Luci ？

1105384898, 2024-10-22:
> 请教：这是选择哪个主题的 Luci ？

argon啊

DanielChen008, 2024-10-22:
> > 请教：这是选择哪个主题的 Luci ？
> 
> argon啊

感谢，我昨晚编译选择的是用了几年的那个 NetGear 主题，刷固件后发现主题崩了，今晚回家用这个主题试试。

1105384898, 2024-10-22:
> > > 请教：这是选择哪个主题的 Luci ？
> > 
> > 
> > argon啊
> 
> 感谢，我昨晚编译选择的是用了几年的那个 NetGear 主题，刷固件后发现主题崩了，今晚回家用这个主题试试。

没事，现在应该只有默认或者argon可以，更新了luci2老的主题不兼容了

Azbuzz, 2024-10-24:
能解决吗

trainSu, 2024-10-25:
同样的问题

coffeelb, 2024-10-30:
同样的问题，但是这个UPNP有生效吗，如果有生效但是不显示也无所谓了

coffeelb, 2024-11-01:
> 没生效

是的，我也测试了没生效

XinSSS, 2024-11-18:
新版本这个插件好像失效了。
不知道会不会修哎😮‍💨

closed 2024-11-20 by coolsnowwolf