coolsnowwolf/lede #12562: 求助如何修改最新编译的默认主题

psonys, 2024-10-20:
### 详细叙述

拉取最新的进行编译，但是默认主题无法通过修改feeds/luci/collections/luci/Makefile进行修改了
现在文件内
LUCI_TITLE:=LuCI interface with Uhttpd as Webserver (default)
LUCI_DESCRIPTION:=Standard OpenWrt set including package management and attended sysupgrades support
LUCI_DEPENDS:= \
	+luci-compat \
	+luci-light \
	+luci-app-opkg
不知道修改哪个了

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

之前的是
LUCI_TITLE:=Standard OpenWrt set including full admin with ppp support and the default Bootstrap theme
LUCI_DEPENDS:= \
	+uhttpd +uhttpd-mod-ubus +luci-mod-admin-full +luci-theme-design \
	+luci-app-firewall +luci-proto-ppp +libiwinfo-lua \
	+rpcd-mod-rrdns

suanfalalala, 2024-10-20:
同问我也想知道

mcusee, 2024-10-20:
同问

lphgor, 2024-10-20:
`luci-light`里面
https://github.com/coolsnowwolf/luci/blob/3c6a9fd719cfdb8319d4c650c2ca7821a6c6bc19/collections/luci-light/Makefile#L19

psonys, 2024-10-20:
已测试，可以使用

closed 2024-10-20 by psonys