coolsnowwolf/lede #12564: mt7925网卡无线编辑错误 TypeError Cannot read properties of undefined (reading 'push')

IDeLoveYou, 2024-10-20:
### 详细叙述

[image]

mt7925网卡点击编辑或者添加都会报错，无线驱动应该是没问题的，手机可以搜到wifi信号

```javascript
TypeError Cannot read properties of undefined (reading 'push')
```

/etc/config/wireless配置如下：

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path '3c0000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option country 'US'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

hinlink-h68k（rockchip-rk3568）

### 详细日志

🈚️

aiamadeus, 2024-10-21:
这个估计纯粹就是新 luci 不兼容

IDeLoveYou, 2024-10-21:
那这个问题该怎么解决呀

IDeLoveYou, 2024-10-21:
> 这个估计纯粹就是新 luci 不兼容

请问这个问题该怎么解决呀，我用的默认的配置

coolsnowwolf, 2024-10-23:
mt7925 额外支持了一些 6Ghz 的信道在目前版本没定义

aiamadeus, 2024-10-23:
然而网卡不支持 6G 也有这个提示...

IDeLoveYou, 2024-10-23:
我临时将feeds.conf.default中的luci切回主分支编译使用，希望能尽早支持吧

closed 2024-10-23 by IDeLoveYou