coolsnowwolf/lede #12565: 23.05 LUCI页面 端口状态显示异常

Matrix-CN, 2024-10-20:
### 详细叙述

固件版本
OpenWrt R24.9.18 (2024-10-20) / LuCI openwrt-23.05 branch git-24.293.38472-3c6a9fd
主页端口状态只显示 eth0 eth1，缺两个
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N5105 4口

### 详细日志

....

Matrix-CN, 2024-10-20:
[image]
[image]

coolsnowwolf, 2024-10-20:
新版 luci 默认只显示你配置在 wan / lan 的 eth，可以根据需要拉进去

closed 2024-10-20 by Matrix-CN

wixxm, 2024-11-14:
> 新版 luci 默认只显示你配置在 wan / lan 的 eth，可以根据需要拉进去

啊我也是遇到同样的情况，这个怎么拉进去呃，我在网口设置里边把eth0（虚拟），eth2，eth3（直通）都配置给了lan 但是还是只显示eth0和eth1（wan）

soimy, 2025-03-16:
同样问题，br-lan配置了eth0 eth2 eth3, 但是端口状态只显示了eth0(lan)和eth1(wan)

wixxm, 2025-03-16:
> 同样问题，br-lan配置了eth0 eth2 eth3, 但是端口状态只显示了eth0(lan)和eth1(wan)

你把虚拟机的虚拟网口设置为e1000e就可以全部显示了

soimy, 2025-03-16:
> > 同样问题，br-lan配置了eth0 eth2 eth3, 但是端口状态只显示了eth0(lan)和eth1(wan)
> 
> 你把虚拟机的虚拟网口设置为e1000e就可以全部显示了

多谢提供线索！