coolsnowwolf/lede #12566: 请教下更新luci后如何更改默认ip

Yicons, 2024-10-21:
### 详细叙述

请教下更新luci2之后编译之前如何更改默认ip啊，下面两个路径都更改过了，但是编译完后还是192.168.1.1，imm的改这两个是有效的
`/package/base-files/files/bin/config_generate`
`"/home/athony/lede/feeds/luci/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js"`

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

none

tonyzzt001, 2024-10-21:
sed -i 's/192.168.1.1/192.168.100.1/g' package/base-files/luci2/bin/config_generate

BenjaminX, 2024-10-21:
同问
/package/base-files/files/bin/config_generate
改这个貌似没有效果了

schen39, 2024-10-21:
> 同问 /package/base-files/files/bin/config_generate 改这个貌似没有效果了

确实没有效果了

dongfangy2011, 2024-10-21:
在quick statrt里的Len settings=>IPv4 Address更改

BenjaminX, 2024-10-21:
> 在quick statrt里的Len settings=>IPv4 Address更改

没听懂呢，具体那里修改？
一般这类不会去 make menuconfig 中修改，会在编译前用脚本去修改。

Yicons, 2024-10-21:
> sed -i 's/192.168.1.1/192.168.100.1/g' package/base-files/luci2/bin/config_generate

实测可以，非常感谢

lpflhh, 2025-06-10:
package/base-files/luci2/bin/config_generate