coolsnowwolf/lede #12569: 新版luci默认配置文件编译之后luci修改配置信息无法保存

wixxm, 2024-10-21:
### 详细叙述

[image]
如题，新版本luci编译，pve安装系统然后再控制台更改ip之后进入openwrt后台然后修改网口配置，保存应用失败。不知道大家有没有遇到过。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

openwrt R24.9.18  / LuCI openwrt-23.05 branch git-24.295.20366-c150a0d

wixxm, 2024-10-21:
[image]
[image]

pangzhaoyhy, 2024-10-21:
新版luci反应很慢,还是老版luci好用

coolsnowwolf, 2024-10-22:
[image]
改网络点击强制应用就好了，JS luci 就这个样子

wixxm, 2024-10-25:
> ![图像](https://private-user-images.githubusercontent.com/31687149/378803665-0fbb458f-2b2b-47de-82ce-28c26c177ff9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4NTYxOTQsIm5iZiI6MTcyOTg1NTg5NCwicGF0aCI6Ii8zMTY4NzE0OS8zNzg4MDM2NjUtMGZiYjQ1OGYtMmIyYi00N2RlLTgyY2UtMjhjMjZjMTc3ZmY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI1VDExMzEzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY3NTk5NDE2OTkxNDIxODkzYTczOGEwNTNmZWYwM2E5NDBkZWQwMzI3NWE0ODRlNjBiMjA0MzY1YTMzYmE0YzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.EvbPpbC6Qxu9pctFZKKNjqduoaxJ-9wn6prcMZraOZo) 改网络点击强制应用就好了，JS luci 就这个样子

好的 大佬 我今天更新一下源码我在编译一下试试，强制应用看看。感谢大佬回复

wixxm, 2024-10-28:
> ![image](https://private-user-images.githubusercontent.com/31687149/378803665-0fbb458f-2b2b-47de-82ce-28c26c177ff9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

[image]
[image]
今天用最新源码编译之后选择强制应用还是失败

wixxm, 2024-10-28:
[image]
[image]
固件信息

coolsnowwolf, 2024-10-28:
别用不兼容的主题

closed 2024-11-20 by coolsnowwolf

Rhuua, 2024-12-09:
> > 不使用不兼容的主题
> 
> 我用argon主题也是一样。。。 我上边截图用的就是argon主题，然后我怕是我编译问题，我又重新编译了一次去掉了argon主题。很奇怪不知道是不是那个插件不兼容嘛？

能否问下问题是否解决？我今天新拉取的仓库编译的树莓派3b固件也遇到了这个问题，具体为烧录后第一次启动系统可以在luci保存配置，重启后就不行了