coolsnowwolf/lede #12586: 新版的DDNS不太正常

qqsir-dev, 2024-10-22:
### 详细叙述

设置DDNS的时候显示没有安装服务
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86 软路由 云编译

### 详细日志

没有编译错误日志

Jason6111, 2024-10-22:
需要手动在安装列表 对吧 我以为是我一人的问题呢

xeyesen, 2024-10-23:
我的也是这样，之前的配置能运行，就是点修改啥都改不了，ddns供应商空的

drw2005, 2024-10-24:
同样的情况，都已正常安装，但是没有DDNS服务。
这个有没有解决办法啊？用不了啦！

Jason6111, 2024-10-24:
> 同样的情况，都已正常安装，但是没有DDNS服务。 这个有没有解决办法啊？用不了啦！

软件列表 刷新 然后搜cloudflare 手动安装下ddns-scripts-cloudflare 先这么用都后续L大修复呗

drw2005, 2024-10-25:
> > 同样的情况，都已正常安装，但是没有DDNS服务。 这个有没有解决办法啊？用不了啦！
> 
> 软件列表 刷新 然后搜cloudflare 手动安装下ddns-scripts-cloudflare 先这么用都后续L大修复呗

谢谢指点。

drw2005, 2024-10-25:
> > 同样的情况，都已正常安装，但是没有DDNS服务。 这个有没有解决办法啊？用不了啦！
> 
> 软件列表 刷新 然后搜cloudflare 手动安装下ddns-scripts-cloudflare 先这么用都后续L大修复呗

发现服务列表里找不到aliyun，明明ddns-scripts_aliyun 1.0.3-1已经安装了。

Jason6111, 2024-10-25:
> > > 同样的情况，都已正常安装，但是没有DDNS服务。 这个有没有解决办法啊？用不了啦！
> > 
> > 
> > 软件列表 刷新 然后搜cloudflare 手动安装下ddns-scripts-cloudflare 先这么用都后续L大修复呗
> 
> 发现服务列表里找不到aliyun，明明ddns-scripts_aliyun 1.0.3-1已经安装了。

我把官方的ddns那个挪过来在测试了

evenwangcn, 2024-10-26:
+1，同样的问题，ddns-scripts没有安装，要手动安装。等L大修复。

closed 2024-10-26 by coolsnowwolf