coolsnowwolf/lede #12589: 编译docker失败，请教各位大佬该如何处理？

dghuig, 2024-10-23:
### 详细叙述

选择编译docker失败，去掉docker后编译没有问题，但是需要docker，请问该如何解决？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AArch64

### 详细日志

ERROR: Expected 'PKG_VERSION:=1.7.22' in '../containerd/Makefile', found 'PKG_VERSION:='
make[3]: *** [Makefile:194: /home/huig/lede/build_dir/target-aarch64_generic_musl/dockerd-27.3.1/.prepared_2b9aaca6bc104c8ed0fa15c317e54ef0_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/huig/lede/feeds/kenzo/dockerd'
time: package/feeds/kenzo/dockerd/compile#0.45#1.65#2.60
    ERROR: package/feeds/kenzo/dockerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/dockerd/compile] Error 1
make[2]: Leaving directory '/home/huig/lede'
make[1]: *** [package/Makefile:110: /home/huig/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/huig/lede'
make: *** [/home/huig/lede/include/toplevel.mk:231：world] 错误 2

xingshu1990, 2024-10-23:
[image]

下午刚编译了一个树莓派3B+环境的  5.15内核版本的固件，带docker
编译顺序：
1.先选择好固件环境（make defconfig前3行的内容），其他默认不做调整，保存config文件。
2.make download -j8（如果前面已经备份过/lede/dl文件夹，复制回来）
make V=s -j5（我是知道我vbox环境给的是5线程），
并且多数情况下编译失败的原因，主要是dl文件的下载不完整导致的，
既然dl已经下载完整，多数情况下直接多线程干他。

大概40分钟-1小时左右，编译成功。
如果不成功的话，先看报错信息，只要不是内核调整导致的补丁没修等问题，或者只要你第一次编译的时候，你没勾选过多的插件，多数情况下都会成功。
第一次成功编译以后，因为要编译斯斯R+，复制执行下面的一键命令（双仓库）版本
https://github.com/kenzok8/small/issues/148
然后make defconfig执行后，进去勾选斯斯R+ 、luci-app-dockerman 、ttyd 、diskman ，勾选了argon主题。
继续：
make download -j8
make V=s -j5

**今天下午我失败的多数原因是**：
1.手动修改/home/wpc/lede/target/linux/【自己需要的架构】/Makefile中的KERNEL_PATCHVER:=，但不知道是否需要修改其他地方，遂直接继续按照L大教程编译，但是出现设备树问题：
```
这个问题主要是因为编译过程中缺少必要的设备树文件 (.dts)，导致构建失败。具体错误信息表明 bcm2710-rpi-3-b-plus.dtb 和 bcm2711-rpi-4-b.dtb 这些设备树二进制文件在生成时出错了：

关键错误信息：
fatal error: No such file or directory
这个错误表示编译器找不到 bcm2711-rpi-4-b.dts 文件。
make: *** Error 2 和 target/linux failed to build
表明整个编译流程中途失败。
```
2.编译的时候，勾选了openwrt支持多种文件系统，结果编译失败。
3.按照网络上的教程：U盘插入识别和自动挂载，另外勾选了USB网卡之类的，编译过程中，源码太大，编译报错。

dghuig, 2024-10-23:
> 这个错误，用chatgpt翻译一下： 这个错误提示说明在编译dockerd包时，../containerd/Makefile中缺少PKG_VERSION的指定。它希望看到你PKG_VERSION:=1.7.22，但实际发现是PKG_VERSION :=（空值）。
> 
> 解决步骤： 检查../containerd/Makefile文件：在lede/feeds/kenzo/containerd/Makefile文件中找到PKG_VERSION这行，确认其值是否被正确设置。如果为空，尝试手动添加正确的版本号，例如：
> 
> PKG_VERSION:=1.7.22
> 
> ![图像](https://private-user-images.githubusercontent.com/18671173/379142742-a66c4bdf-c855-46d7-8992-63199d15e870.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2NzAyNzUsIm5iZiI6MTcyOTY2OTk3NSwicGF0aCI6Ii8xODY3MTE3My8zNzkxNDI3NDItYTY2YzRiZGYtYzg1NS00NmQ3LTg5OTItNjMxOTlkMTVlODcwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIzVDA3NTI1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0NGY0ZGY5ZjRmNzdmZWU4MWFlYjk0MDk4NjlhNzdmYzVkNzgxZjhjY2NlMzdjYzBlOWEzYjdhMDI2MGU1N2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.S08MFCcnaTMW1VcQn9Ua9PPslokUWgndICFogQ3xNEc)
> 
> ## 先说结论：编译的时候，勾选luci-app-dockerman
> 下午刚编译了一个树莓派3B+环境的5.15内核版本的固件，带docker 编译顺序： 1.先选择好的固件环境（make defconfig前3行的内容），其他默认不做调整，保存config文件。 2 .make download -j8（如果前面已经备份过/lede/dl文件夹，复制回来） make V=s -j5（我是知道我vbox环境给5的是线程）， 并且大多数情况下编译失败的原因，主要是dl文件的下载不完整导致的， 既然dl已经下载完整，大多数情况下直接多线程干他。
> 
> 大概40分钟-1小时左右，编译成功。 如果不成功的话，先看报错信息，只要不是内核调整导致的补丁没修等问题， 或者只要你第一次编译的时候，你没有勾选过多的插件，大多数情况下都能成功。
> 
> 下载错误：系统编译过程中需要用依赖库（包）没下载完成，或者原始链接失效。 据我了解，会下载linux 5.1* linux 6.1* linux 6.6*其中一个或者多个文件，而那个下载过程中就只有几个KB， 遂CTRL+C，自行下载相应的依赖文件，保存到相应位置。
> 
> 第一次成功编译以后，因为要编译ssR+，复制执行下面的一键命令（双仓库）版本 [kenzok8/small#148](https://github.com/kenzok8/small/issues/148)
> 
> 然后make defconfig执行后，勾选斯斯R+、luci-app-dockerman、ttyd、diskman，勾选了argon主题。 继续： make download -j8 make V=s -j5
> 
> **今天下午我失败的大部分原因是**： 1.手动修改/home/wpc/lede/target/linux/【自己需要的架构】/Makefile中的KERNEL_PATCHVER:=， 但不知道是否需要修改其他位置，遂直接继续根据L大教程编译，但是出现设备树问题：
> 
> ```
> 这个问题主要是因为编译过程中缺少必要的设备树文件 (.dts)，导致构建失败。
> 具体错误信息表明 bcm2710-rpi-3-b-plus.dtb 和 bcm2711-rpi-4-b.dtb 这些设备树二进制文件在生成时出错了：
> 
> 关键错误信息：
> fatal error: No such file or directory
> 这个错误表示编译器找不到 bcm2711-rpi-4-b.dts 文件。
> make: *** Error 2 和 target/linux failed to build
> 表明整个编译流程中途失败。
> ```
> 
> 2.编译的时候，勾选了openwrt支持多个文件系统，结果编译失败。 3.按照网络上的教程：U盘插入识别和自动挂载，另外勾选了USB占用之类的，编译过程中，文件繁琐，硬盘容量庞大，编译报错。

在lede/feeds/kenzo/containerd路径没有找到containerd！！！在lede/feeds/packages/utils底下有containerd，但里的Makefile中没有缺少PKG_VERSION的指定值，PKG_VERSION :=1.7.18

xingshu1990, 2024-10-23:
> 在lede/feeds/kenzo/containerd路径没有找到containerd！！！在lede/feeds/packages/utils底下有containerd，但里的Makefile中没有缺少PKG_VERSION的指定值，PKG_VERSION :=1.7.18

那就等大佬了，我不过来瞎说了。

closed 2024-10-23 by dghuig