coolsnowwolf/lede #12597: 云编译失败

lexin8, 2024-10-25:
### 详细叙述

master版本，之前master不是luci23.05的时候还是正常的，最近就一直过不去

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xiaomi_redmi-router-ac2100

### 详细日志

`4 thread compile
make[2]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
 make[1] world
 make[2] package/cleanup
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/config'
make[1]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
make[2]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
make[3]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux'
make[4]: Entering directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux/ramips'
rm -f /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/localversion
/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/kconfig.pl  + /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux/generic/config-5.4 /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux/ramips/mt7621/config-5.4 > /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/.config >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target
/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/tmp/.packageinfo /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/.config 5.4 > /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.override
/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.target /dev/null /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.override > /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set
mv /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.old
grep -v INITRAMFS /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.old > /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set
rm -rf /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/modules
cmp -s /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.prev || { cp /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config; cp /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.prev; }
export MAKEFLAGS= ; [ -d /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/user_headers ] || make -C /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284 KCFLAGS="-fmacro-prefix-map=/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Oct 24 14:16:05 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.284 CC="mipsel-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/user_headers headers_install
grep '=[ym]' /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.config.set | LC_ALL=C sort | /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/host/bin/mkhash md5 > /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.vermagic
touch /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/.configured
rm -f /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/vmlinux /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.284/System.map
[... cut ...]
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
make[1]: *** [target/Makefile:23: /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
make: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.`

sxml2015, 2024-10-25:
luci 23.05 云编译一直超时 。换回master又正常编译 日志又没找到Error. 跟 feeds.conf.default 源有关吗？

oppen321, 2024-10-25:
把Action的Caches缓存清理了重新编译应该就好了

closed 2024-10-26 by lexin8