coolsnowwolf/lede #12604: 修复DDNS后，提示确实ipkg

evenwangcn, 2024-10-27:
### 详细叙述

[
[image]
](url)

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/controller/ddns.lua:14: module 'luci.model.ipkg' not found:
no field package.preload['luci.model.ipkg']
no file './luci/model/ipkg.lua'
no file '/usr/share/lua/luci/model/ipkg.lua'
no file '/usr/share/lua/luci/model/ipkg/init.lua'
no file '/usr/lib/lua/luci/model/ipkg.lua'
no file '/usr/lib/lua/luci/model/ipkg/init.lua'
no file './luci/model/ipkg.so'
no file '/usr/lib/lua/luci/model/ipkg.so'
no file '/usr/lib/lua/loadall.so'
no file './luci.so'
no file '/usr/lib/lua/luci.so'
no file '/usr/lib/lua/loadall.so'

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function build_pagetree (/usr/share/ucode/luci/dispatcher.uc:382:73)
  called from function menu_json (/usr/share/ucode/luci/dispatcher.uc:439:26)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:898:24)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

evenwangcn, 2024-10-27:
/package/lean/default-settings/files/zzz-default-settings文件替换openwrt_luci的镜像源的是不是要替换成更高的版本
sed -i '/openwrt_luci/ { s/snapshots/releases\/18.06.9/g; }'  /etc/opkg/distfeeds.conf
我看18.06.9版本中并没有luci-lib-ipkg包

evenwangcn, 2024-10-27:
luci-app-ddns的Makefile里面的依赖是不是要加上+luci-lib-ipkg

coolsnowwolf, 2024-10-27:
修复了

closed 2024-10-27 by evenwangcn