coolsnowwolf/lede #12610: EA8500编译报错，应该是修改Shadowsocks Client Selection组件所致

Hirro029, 2024-10-28:
### 详细叙述

感觉应该是修改Shadowsocks Client Selection由Shadowsocks-libev改为Shadowsocks-rust所致。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

EA8500

### 详细日志

make[3]: Entering directory '/opt/lede/feeds/packages/lang/rust'
. /opt/lede/include/shell.sh; /opt/lede/staging_dir/host/bin/libdeflate-gzip -dc /opt/lede/dl/rustc-1.80.1-src.tar.gz | tar -C /opt/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.80.1-src/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:111: /opt/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.80.1-src/.prepared4eece2552e829dd1fc3b7fd15ce8bfec_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/opt/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#3.63#6.04#25.03
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2024-10-28:
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors

coolsnowwolf, 2024-10-28:
被墙了，删除 dl 目录再来一次

Hirro029, 2024-10-28:
make[3]: Entering directory '/opt/lede/feeds/packages/lang/rust'
mkdir -p /opt/lede/dl
SHELL= flock /opt/lede/tmp/.rustc-1.80.1-src.tar.gz.flock -c '  	/opt/lede/scripts/download.pl "/opt/lede/dl" "rustc-1.80.1-src.tar.gz" "2c0b8f643942dcb810cbcc50f292564b1b6e44db5d5f45091153996df95d2dc4" "" "https://static.rust-lang.org/dist/"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://static.rust-lang.org/dist/rustc-1.80.1-src.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  360M  100  360M    0     0  1733k      0  0:03:32  0:03:32 --:--:-- 1758k
. /opt/lede/include/shell.sh; /opt/lede/staging_dir/host/bin/libdeflate-gzip -dc /opt/lede/dl/rustc-1.80.1-src.tar.gz | tar -C /opt/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.80.1-src/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:111: /opt/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.80.1-src/.prepared31f848a6b80d1d2e44f9db1d15cf4254_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/opt/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#9.00#9.33#239.78
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231：world] 错误 2

挂上全局代理，还是报这个错。

coolsnowwolf, 2024-10-28:
tar: This does not look like a tar archive 还是被墙了

Hirro029, 2024-10-29:
试了一下，不管怎么下载下来的rustc-1.80.1-src.tar.gz这个文件，用不用代理都是报这个错。
能不能再提供一下别的下载点？

Hirro029, 2024-10-31:
试了一下Luci选择23.05，能正常编译过去，并且rustc-1.80.1-src.tar.gz的文件是一样的。
文件: C:\Users\Administrator\Desktop\rustc-1.80.1-src.tar (3).gz
大小: 377927214 字节
修改时间: 2024年10月30日, 18:04:41
MD5: 043F14D8A15BDB8C7B21D67EB5E468F7
SHA1: 7B10AEC3E0595197B14C7C29F6618127691395F3
CRC32: 2BB1131D

closed 2024-11-09 by coolsnowwolf

justamanm, 2024-11-11:
问下Luci怎么选择23.05，我也遇到这个问题了

Hirro029, 2024-11-11:
> 问下Luci怎么选择23.05，我也遇到这个问题了

修改 feeds.conf.default这个文件
#src-git luci https://github.com/coolsnowwolf/luci
src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05
这两行改成这样就是使用23.05

justamanm, 2024-11-12:
> > 问下Luci怎么选择23.05，我也遇到这个问题了
> 
> 修改 feeds.conf.default这个文件 #src-git luci https://github.com/coolsnowwolf/luci src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05 这两行改成这样就是使用23.05

好的，谢谢，我重新编译下

Hirro029, 2024-11-13:
故障重现和解决方法：
[image]
[image]
[image]
看了一下libdeflate相关文档，发现是该命令无法解压缩rustc的文件，导致二次解压tar失败。
后来发现是虚拟机内存设置过小造成的。libdeflate解压rustc文件超过2G，所以编译设备内存需要超过2G。
后面截图已经可以正常编译。
这个报错还有一个可能就是L大说的，因为被墙导致下载数据的hash值和正式文件不一致。