coolsnowwolf/lede #12614: 树莓派4B刷入之后web界面报错无法进入

rex5652, 2024-10-28:
### 详细叙述

在 e0f6d6b7ead7abd2d08c1056d6f5a5a9b9c3ecac 这个commit编译之后刷入固件报错。
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template themes/argon/header'.
A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '_entries' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:68: in function 'node'
[string "/usr/lib/lua/luci/view/themes/argon/header...."]:20: in main chunk
In error0, file [C]
called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function [anonymous function) (/usr/lib/lua/luci/ucodebridge.lua:37)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
called fron function [arrow function) (/usr/share/ucode/luci/runtime.uc:141:63)
called from function render ([C])
called from function [anonymous function) (/usr/share/ucode/luci/runtime.uc:141:64)
called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
called from anonymous function (/wr/cgi-bin/luci:39:12)
die(ex)
Near here

查了说是theme的问题，重装theme进去之后感觉有各种bug，比如wireguard接口就没了。opkg也装不上说和内核版本不兼容。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

树莓派4B

### 详细日志

[image]

coolsnowwolf, 2024-10-28:
别用 lua 版本的主题

xingshu1990, 2024-10-29:
ipk如果是自己编译，但是安装不上，我个人觉得可以这样：
现在L大这边大版本是5.15以前，5.15，5.4，6.1，6.6，
我不知道6.1 5.4是否有适配，所以不举例。

假定编译5.15.148版本固件+插件：
这两个都会有.148的数据
https://github.com/coolsnowwolf/lede/commits/master/include/kernel-5.15
[image]

源码根目录(一般是lede)/include/kernel-5.15（对应版本号，当前以5.15小版本举例），当下先不管这个，

[image]

下面说下出处。
[image]

网址：**https://github.com/coolsnowwolf/lede/commit/72154bce76e591a07523270407e819baea6fcd6e**

[image]

先在源码根目录/include/kernel-version.mk输入：
LINUX_VERSION-5.15 = .148
LINUX_KERNEL_HASH-5.15.148 = c48575c97fd9f4767cbe50a13b1b2b40ee42830aba3182fabd35a03259a6e5d8

我没按github上的这个位置来，直接在最顶端粘贴。
[image]

树莓派3B+是BCM2710，因此我的路径是这个，4B或其他，需要自行找目录。
然后查看Makefile文件的KERNEL_PATCHVER:=是多少。其他不动
[image]

然后make dirclean
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make download -j8
make V=s -j1

如果以上有哪个地方有问题，麻烦告知错误。

closed 2024-10-31 by coolsnowwolf