coolsnowwolf/lede #12615: 新的DDNS cloudflara更新失败

Jason6111, 2024-10-28:
### 详细叙述

新的DDNS cloudflara发送更新失败 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

 045423       : verbose mode  : 0 - run normal, NO console output
 045423       : check interval: 600 seconds
 045423       : force interval: 4320 seconds
 045423       : retry interval: 60 seconds
 045423       : retry max count : 0 times
 045423       : No old process
 045423       : last update: never
 045423       : Detect registered/public IP
 045423       : #> /usr/bin/host -t A www.j*****g  >/var/run/ddns/cloudflare.dat 2>/var/run/ddns/cloudflare.err
 045423       : Registered IP '110.****.35.129' detected
 045423  info : Starting main loop at 2024-10-29 04:54
 045423       : Detect current IP on 'network'
 045423       : Current IP '110.***.200.221' detected on network 'wan'
 045423       : Update needed - L: '110.***.200.221' <> R: '110.***.35.129'
 045423       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 045423       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflare.dat --stderr /var/run/ddns/cloudflare.err --noproxy '*' --header 'X-Auth-Email: j*****n'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=www.j******g'
 045424  WARN : Could not detect 'zone id' for domain.tld: 'www.j*****g'
 045424 ERROR : No update send to DDNS Provider

coolsnowwolf, 2024-10-29:
cloudflare 域名的 zone_id 找不到，这个我没在用，你找找官方

Jason6111, 2024-10-29:
> cloudflare 域名的 zone_id 找不到，这个我没在用，你找找官方

感谢L大提点 暂时提取到原来的替换了
# ddns cloudflare
curl -o new_update_cloudflare_com_v4.sh https://raw.githubusercontent.com/Jason6111/Openwrt_Beta/refs/heads/main/patch/update_cloudflare_com_v4.sh
mv new_update_cloudflare_com_v4.sh ./feeds/packages/net/ddns-scripts/files/usr/lib/ddns/update_cloudflare_com_v4.sh
目前看工作正常

closed 2024-10-29 by Jason6111