coolsnowwolf/lede #12616: 新版本SoftEtherVPN相关问题

wi11iamzhao, 2024-10-28:
### 详细叙述

更新了新版本（大约从R24.9.9->当前最新）后，LUCI界面中的softhervpn开关页面消失（原本在/cgi-bin/luci/admin/vpn/softethervpn），状态页面中多出来了一个SoftEther Status，但是打开始终显示Loading account information…
编译时已经执行过：
`
rm -rf .config   
make menuconfig
`
并且在LUCI->Application中重新选择了luci-app-softether。
系统进程中可以看到/var/softethervpn/vpnbridge  /var/softethervpn/vpnserver /var/softethervpn/vpnclient ，SoftEtherVPN的管理工具也能正常连上。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86-64软路由,ESXI虚拟化

### 详细日志

Tue Oct 29 05:30:59 2024 user.notice softethervpn5: Starting softether bridge service.
Tue Oct 29 05:30:59 2024 daemon.err vlmcsd[7946]: Warning: Can't read /etc/vlmcsd.ini: No such file or directory
Tue Oct 29 05:30:59 2024 daemon.err vlmcsd[7946]: Warning: Can't read /etc/vlmcsd.ini: No such file or directory
Tue Oct 29 05:30:59 2024 user.notice firewall: Reloading firewall due to ifup of lan_10gb_2 (eth5)
Tue Oct 29 05:30:59 2024 user.notice softethervpn5: Starting softether vpnserver service.
Tue Oct 29 05:30:59 2024 user.notice softethervpn5: Starting softether client service.
Tue Oct 29 05:30:59 2024 daemon.err mosdns[8143]: 2024-10-28T21:30:59.633Z	INFO	working directory changed	{"path": "/etc/mosdns"}
Tue Oct 29 05:30:59 2024 daemon.err mosdns[8143]: 2024-10-28T21:30:59.634Z	INFO	main config loaded	{"file": "/etc/mosdns/config_custom.yaml"}
Tue Oct 29 05:30:59 2024 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan_10gb_2 (eth5)
Tue Oct 29 05:30:59 2024 user.notice dnsmasq: found already running DHCP-server on interface 'eth1' refusing to start, use 'option force 1' to override
Tue Oct 29 05:31:00 2024 daemon.notice softethervpn-server[8018]: Writing vpn_server.config (AutoSaveConfigSpan set to < 86400 > seconds)
Tue Oct 29 05:31:00 2024 daemon.notice softethervpn-server[8437]: Writing vpn_server.config (AutoSaveConfigSpan set to < 86400 > seconds)
Tue Oct 29 05:31:00 2024 daemon.err collectd[6819]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/openwrt/memory/memory-cached.rrd: Not enough arguments

wi11iamzhao, 2024-10-30:
补充：由于在系统升级前已经设置并启用SoftEtherVPN，目前虽然无法显示状态但是工作正常，内外网均可正常连接（除了证书被自动重置成了自签名证书）

sgx800, 2024-11-02:
确实有这个问题，UI显示不正常

coolsnowwolf, 2024-11-07:
mark

closed 2024-11-20 by coolsnowwolf

wldonglei, 2024-12-14:
我也是遇到了 我是编译固件新装的，版本号LEDE R24.11.11 / LuCI openwrt-23.05 branch git-24.326.41750-70addca，找不到开关入口，管理工具也连不上

gitHardie, 2025-02-28:
不影响使用,目前看 这个status是无法使用的,rpcd权限问题  Access to command denied by ACL

zl810881283, 2025-03-02:
我也遇到这个问题

zl810881283, 2025-03-15:
已经收到您的来信，我会尽快回复。