coolsnowwolf/lede #12617: 最新编译的SSR无法启动

koogg, 2024-10-29:
### 详细叙述

刚编译了最新的固件，ssr启动一直启动不成功，日志显示
2024-10-29 08:00:14: ssrplus redir tcp error.restart!

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X64

### 详细日志

2024-10-29 08:00:14: ssrplus redir tcp error.restart!

koogg, 2024-10-29:
> 这个错误通常与 SSR Plus 的配置或网络设置有关。 先看看dns设置是否正常， 另外节点是否有速度。 SSR中的运行模式、和DNS 解析方式是怎么样。
> 
> 据我稀疏的测试，如果是选择GFW，SSR报错说是找不到gfw文件， 但是我在对应的文件夹里能找到。 另外部分人可能还有开启smartdns搭配使用的情况，先取消smartdns单独测试ssr是否运行正常。

配置和网络应该都没问题，因为我换回上一个版本后一切正常，什么配置都没有改动服务就自动启动了。  我也没有使用smartdns服务，我只安装了 SSR+ DDNS+upnp，其他都没有安装

xingshu1990, 2024-10-29:
> > 这个错误通常与 SSR Plus 的配置或网络设置有关。 先看看dns设置是否正常， 另外节点是否有速度。 SSR中的运行模式、和DNS 解析方式是怎么样。
> > 据我稀疏的测试，如果是选择GFW，SSR报错说是找不到gfw文件， 但是我在对应的文件夹里能找到。 另外部分人可能还有开启smartdns搭配使用的情况，先取消smartdns单独测试ssr是否运行正常。
> 
> 配置和网络应该都没问题，因为我换回上一个版本后一切正常，什么配置都没有改动服务就自动启动了。 我也没有使用smartdns服务，我只安装了 SSR+ DDNS+upnp，其他都没有安装

那得去软件仓库那边，因为这边是L大的源码仓库，已经剥离了ssr了

jiaopengzi, 2024-10-29:
遇到相同的问题`ssrplus redir tcp error.restart!`,一直重启。,好在原来的固件留存了，回滚回去。新版本的稳定性再等等吧。

```
2024-10-29 09:56:46: -----------end------------
2024-10-29 09:56:41: Main node:Xray 24.10.16 (Xray, Penetrates Everything.) OpenWrt (go1.23.1 linux/amd64) Started!
2024-10-29 09:56:41: ----------start------------
2024-10-29 09:56:37: ssrplus redir tcp error.restart!
2024-10-29 09:56:07: -----------end------------
2024-10-29 09:56:02: Main node:Xray 24.10.16 (Xray, Penetrates Everything.) OpenWrt (go1.23.1 linux/amd64) Started!
2024-10-29 09:56:02: ----------start------------
2024-10-29 09:55:59: ssrplus redir tcp error.restart!
2024-10-29 09:55:29: -----------end------------
2024-10-29 09:55:24: Main node:Xray 24.10.16 (Xray, Penetrates Everything.) OpenWrt (go1.23.1 linux/amd64) Started!
```

Jason6111, 2024-10-29:
先停止 把/etc/config/ss那个删了 重启机器 重新配置下

xingshu1990, 2024-10-29:
树莓派3B+ 固件版本24.2.2版本 内核5.15.148
把原先的正在运行的SSR删除了，把今天自行编译的SSR给上传安装。
运行正常。

coolsnowwolf, 2024-10-29:
xray 版本更新了，接口改了，整套都要更新到新版

hcym, 2024-10-29:
总算有官方声明
知道了，谢谢大佬

koogg, 2024-10-29:
> xray 版本更新了，接口改了，整套都要更新到新版

感谢大佬回复，我是直接还原配置，重新配置解决的问题，不过节点那里，上移和下移，界面显示的是 在线和下线，可以调整一下，以免误解意思。

Jason6111, 2024-10-29:
> > xray 版本更新了，接口改了，整套都要更新到新版
> 
> 感谢大佬回复，我是直接还原配置，重新配置解决的问题，不过节点那里，上移和下移，界面显示的是 在线和下线，可以调整一下，以免误解意思。 ![PixPin_2024-10-29_17-00-47](https://private-user-images.githubusercontent.com/45394398/381052540-3926160c-91a0-49bf-a66b-5df883034ae5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxOTUxODIsIm5iZiI6MTczMDE5NDg4MiwicGF0aCI6Ii80NTM5NDM5OC8zODEwNTI1NDAtMzkyNjE2MGMtOTFhMC00OWJmLWE2NmItNWRmODgzMDM0YWU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI5VDA5NDEyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIwNTQxYzZhZWM3M2Q2NTAzZDA0MWM2ZDg5OTRlYWE5ODMxOWVlNmZjY2UyNmIyMDhlOGZiZDJiYjdhZGQ0ZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.G3BIBocO8ghdFaaDXPEQCj6-BR0634soEyaJekgvnqY)

中午就告诉过你方法了 哈哈哈 可以关闭了

koogg, 2024-10-29:
> 中午就告诉过你方法了 哈哈哈 可以关闭了

我在etc下没有ss文件，只有shadowsocksr，我直接给删了文件，，，然后发现配置完全没法配了，才重置的配置重新来一波

Jason6111, 2024-10-29:
> 我在etc下没有ss文件，只有shadowsocksr，我直接给删了文件，，，然后发现配置完全没法配了，才重置的配置重新来一波

[image]
哈哈哈 我的错没说明白 先停止 删这个然后重启 重启后无法配置再更新下估计 在配置就可以了

jiaopengzi, 2024-10-30:
> xray 版本更新了，接口改了，整套都要更新到新版

看到 Lean 大佬的回复是全套都要变，主要针对的是 ssrp。
我的操作方式是：
1. 备份配置到本地。
2. 使用最新编译固件，不保存配置 刷写固件。
3. ssh 使用默认的ip 192.168.1.1 默认用户root，默认密码password 连接
执行备份默认最新的配置 
`cp /etc/config/shadowsocksr /etc/config/shadowsocksr.new`
`cp -r /etc/ssrplus  /etc/ssrplus.new`
4. 将最新的默认配置备份好以后，在web页面，将第1步的备份配置上传，此时 ssrp 的配置会被覆盖，因为全套使用最新的，所以需要用最新的配置
5. 将第3步备份好的内容在重新覆盖下，此时使用你自己原来的账户密码登录ssh执行
`cp /etc/config/shadowsocksr.new /etc/config/shadowsocksr`
`cp -r /etc/ssrplus.new  /etc/ssrplus`
6. 重启一下机器，在web页面重新配置一下 ssrp 可以快乐的玩耍了。

wi11iamzhao, 2024-10-30:
简单一点的解决方法：  
备份/etc/config/shadowsocksr和/etc/ssrplus到本地，  
然后卸载掉luci-app-ssr-plus_188_all.ipk，  
然后删除/etc/config/shadowsocksr和/etc/ssrplus。  
随后重新安装luci-app-ssr-plus_188_all.ipk，或者再次刷写新的固件。  
这样配置文件就会更新，再手动填写/恢复配置就能正常运行。

closed 2024-11-08 by coolsnowwolf

nebulite, 2025-02-15:
> 简单一点的解决方法： 备份/etc/config/shadowsocksr和/etc/ssrplus到本地， 然后卸载掉luci-app-ssr-plus_188_all.ipk， 然后删除/etc/config/shadowsocksr和/etc/ssrplus。 随后重新安装luci-app-ssr-plus_188_all.ipk，或者再次刷写新的固件。 这样配置文件就会更新，再手动填写/恢复配置就能正常运行。

我重新格式化之后全新刷的固件，重新配置ssrp，但是也是报错 ssrplus redir tcp error.restart!