coolsnowwolf/lede #12619: orange pi 5 plus 编译时报错

qq761861, 2024-10-29:
### 详细叙述

编译报错 orange pi 5 plus 报错信息如下

/home/user/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/ccxUjVL1.ltrans2.ltrans.o: in function `ap_sta_add':
<artificial>:(.text+0x3a14): undefined reference to `sta_track_claim_taxonomy_info'
/home/user/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text+0x3a20): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:1402: hostapd] Error 1
make[4]: Leaving directory '/home/user/lede/build_dir/target-aarch64_generic_musl/hostapd-full-internal/hostapd-2022-07-29-b704dc72/hostapd'
make[3]: *** [Makefile:790: /home/user/lede/build_dir/target-aarch64_generic_musl/hostapd-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/user/lede/package/network/services/hostapd'
time: package/network/services/hostapd/full-internal/compile#20.89#2.50#3.82
    ERROR: package/network/services/hostapd failed to build (build variant: full-internal).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231: world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

orange pi 5 plus

### 详细日志

/home/user/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/ccxUjVL1.ltrans2.ltrans.o: in function `ap_sta_add':
<artificial>:(.text+0x3a14): undefined reference to `sta_track_claim_taxonomy_info'
/home/user/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: <artificial>:(.text+0x3a20): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:1402: hostapd] Error 1
make[4]: Leaving directory '/home/user/lede/build_dir/target-aarch64_generic_musl/hostapd-full-internal/hostapd-2022-07-29-b704dc72/hostapd'
make[3]: *** [Makefile:790: /home/user/lede/build_dir/target-aarch64_generic_musl/hostapd-full-internal/hostapd-2022-07-29-b704dc72/.built] Error 2
make[3]: Leaving directory '/home/user/lede/package/network/services/hostapd'
time: package/network/services/hostapd/full-internal/compile#20.89#2.50#3.82
    ERROR: package/network/services/hostapd failed to build (build variant: full-internal).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231: world] Error 2
[image]

coolsnowwolf, 2024-10-29:
解决了

话说为何要选上这个支持 full ？一般都是用 wpad 吧？

qq761861, 2024-10-29:
用的其他模板自带的，取消后就可以正常编译了

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月29日 15:46 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] hostapd-full-internal 编译时报错 (Issue #12619) |

解决了

话说为何要选上这个支持 full ？一般都是用 wpad 吧？

closed 2024-10-29 by coolsnowwolf