coolsnowwolf/lede #12627: 编译失败

gssiwmd, 2024-10-30:
### 详细叙述

今天更新后，编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:88: warning: addprefix --directory ,$(CODE_COVERAGE_DIRECTORY: non-POSIX variable name
aminclude_static.am:88: (probably a GNU make extension)
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:90: warning: addprefix --prefix ,$(CODE_COVERAGE_DIRECTORY: non-POSIX variable name
aminclude_static.am:90: (probably a GNU make extension)
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:100: warning: ':='-style assignments are not portable
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:100: warning: AM_DISTCHECK_CONFIGURE_FLAGS was already defined in condition TRUE, which includes condition CODE_COVERAGE_ENABLED ...
Makefile.am:167:   'aminclude_static.am' included from here
Makefile.am:21: ... 'AM_DISTCHECK_CONFIGURE_FLAGS' previously defined here
doc/Makefile.am: error: required file 'build-aux/depcomp' not found
doc/Makefile.am:   'automake --add-missing' can install 'depcomp'
make[4]: *** [Makefile:2258: Makefile.in] Error 1
make[4]: Leaving directory '/home/gswangmd/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
make[3]: *** [Makefile:270: /home/gswangmd/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.built] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#32.90#2.97#37.17
    ERROR: package/feeds/packages/gnutls failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@ubunntuserver1:~/lede$

akihara-sam, 2024-10-30:
同样 编译 失败错误相同。 我是 mt7621 。
chatgpt 回答如下：“这个错误主要来源于 gnutls 包在编译过程中找不到 Automake 所需的 build-aux/depcomp 文件。这个文件是依赖项管理的辅助脚本，通常由 Automake 自动生成并放置在 build-aux 目录中。“

limymy, 2024-10-30:
遇到了同样的问题，x86_64架构固件

gotodragon, 2024-10-30:
也是同样的失败

closed 2024-10-30 by coolsnowwolf

gssiwmd, 2024-10-30:
如何解决的呢？

justlovemaki, 2024-10-31:
如何解决的呢？