coolsnowwolf/lede #12628: ipq40xx系列设备在更换的23.05新版本中5G WIFI AC模式信道丢失问题

YEYEv50, 2024-10-30:
### 详细叙述

如图所示，在全面更换的默认LUCI-23.05版本中，5G WIFI AC出现丢失信道的情况，切换到N模式又正常出现信道。在尝试了IMM或者官方OP固件中无此情况。新编译固件，刷入后设置全默认无改动
[image]

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AUSU-RT-AC42U(RT-ACRH17)

### 详细日志

NO

Hirro029, 2024-10-31:
这个应该不是ipq40xx系列的问题，我的ipq806x，型号EA8500也是这个问题。
我看之前还有一个提问的是小米R3G路由也是这个问题。

YEYEv50, 2024-11-01:
> 这个应该不是ipq40xx系列的问题，我的ipq806x，型号EA8500也是这个问题。 我看之前还有一个提问的是小米R3G路由也是这个问题。

这样啊，那不知道啥问题了。退回18.06LUCI又正常

ashaaf, 2024-11-09:
碰见了一样的问题，先前编译的包没发现，具体是在一次修改操作后出现的
[image]
[image]

YEYEv50, 2024-11-11:
> 碰见了一样的问题，先前编译的包没发现，具体是在一次修改操作后出现的 [...]

试了一下全默认编译也是一样的问题，就是只选了型号直接保存

coolsnowwolf, 2024-11-12:
最好把系统日志上传下

Hirro029, 2024-11-13:
系统日志
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.192738] qcom-pcie 1b500000.pci: host bridge /soc/pci@1b500000 ranges:
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.192772] qcom-pcie 1b500000.pci: Parsing ranges property...
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.192826] qcom-pcie 1b500000.pci:       IO 0x000fe00000..0x000fe0ffff -> 0x0000000000
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.192871] qcom-pcie 1b500000.pci:      MEM 0x0008000000..0x000fdfffff -> 0x0008000000
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419413] qcom-pcie 1b500000.pci: Link up
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419595] qcom-pcie 1b500000.pci: PCI host bridge to bus 0000:00
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419624] pci_bus 0000:00: root bus resource [bus 00-ff]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419645] pci_bus 0000:00: root bus resource [io  0x0000-0xffff]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419667] pci_bus 0000:00: root bus resource [mem 0x08000000-0x0fdfffff]
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.419688] pci_bus 0000:00: scanning bus
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419753] pci 0000:00:00.0: [17cb:0101] type 01 class 0x060400
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419914] pci 0000:00:00.0: supports D1
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.419933] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.419958] pci 0000:00:00.0: PME# disabled
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.423504] pci_bus 0000:00: fixups for bus
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.423529] PCI: bus0: Fast back to back transfers disabled
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.423554] pci 0000:00:00.0: scanning [bus 01-ff] behind bridge, pass 0
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.423696] pci_bus 0000:01: scanning bus
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.424171] pci 0000:01:00.0: [168c:0040] type 00 class 0x028000
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.424492] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.425968] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.426027] pci 0000:01:00.0: PME# disabled
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.426431] pci 0000:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.429920] pci_bus 0000:01: fixups for bus
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.430012] PCI: bus1: Fast back to back transfers disabled
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.430034] pci_bus 0000:01: bus scan returning with max=01
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.430053] pci 0000:00:00.0: scanning [bus 01-ff] behind bridge, pass 1
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.430073] pci_bus 0000:00: bus scan returning with max=ff
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.430111] pci 0000:00:00.0: BAR 8: assigned [mem 0x08000000-0x081fffff]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.430139] pci 0000:01:00.0: BAR 0: assigned [mem 0x08000000-0x081fffff 64bit]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.430304] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.430334] pci 0000:00:00.0:   bridge window [mem 0x08000000-0x081fffff]
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.430546] pcieport 0000:00:00.0: assign IRQ: got 44
Wed Nov 13 20:22:45 2024 kern.info kernel: [    0.431427] pcieport 0000:00:00.0: AER: enabled with IRQ 45
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431553] pcieport 0000:00:00.0: saving config space at offset 0x0 (reading 0x10117cb)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431570] pcieport 0000:00:00.0: saving config space at offset 0x4 (reading 0x100547)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431584] pcieport 0000:00:00.0: saving config space at offset 0x8 (reading 0xff000000)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431597] pcieport 0000:00:00.0: saving config space at offset 0xc (reading 0x10010)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431611] pcieport 0000:00:00.0: saving config space at offset 0x10 (reading 0x0)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431625] pcieport 0000:00:00.0: saving config space at offset 0x14 (reading 0x0)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431640] pcieport 0000:00:00.0: saving config space at offset 0x18 (reading 0xff0100)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431653] pcieport 0000:00:00.0: saving config space at offset 0x1c (reading 0xf0)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431667] pcieport 0000:00:00.0: saving config space at offset 0x20 (reading 0x8100800)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431680] pcieport 0000:00:00.0: saving config space at offset 0x24 (reading 0x1fff1)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431694] pcieport 0000:00:00.0: saving config space at offset 0x28 (reading 0x0)
Wed Nov 13 20:22:45 2024 kern.debug kernel: [    0.431707] pcieport 0000:00:00.0: saving config space at offset 0x2c (reading 0x0)
[... cut ...]
[   32.969760] ath: regdomain 0x809c dynamically updated by user
[   33.284934] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   33.285100] br-lan: port 2(wlan1) entered blocking state
[   33.290372] br-lan: port 2(wlan1) entered forwarding state
[   35.543072] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.543350] br-lan: port 3(wlan0) entered blocking state
[   35.548616] br-lan: port 3(wlan0) entered forwarding state
[   40.493745] ath10k_pci 0000:01:00.0: htt tx: fixing invalid VHT TX rate code 0xff

[image]

YEYEv50, 2024-11-13:
> 最好把系统日志上传下

已更新日志，刷入后无更改设置。大佬请查看一下。

Hirro029, 2024-11-14:
小米R3G路由5gWiFi无法启用 #12430
这个好像是一样的问题

SayHi1314, 2024-11-19:
我也是

venpong, 2024-11-19:
俺也一样，刷机后有wifi，修改设置后就无法启动5Gwifi了，也是acrh17

YEYEv50, 2024-11-21:
> 俺也一样，刷机后有wifi，修改设置后就无法启动5Gwifi了，也是acrh17

用WinSCP进入路由器/etc/config 手动编辑wireless文件。参考这样
option channel '44'
option band '5g'
option htmode 'VHT80'
option country 'CN'
option disabled '0'
option cell_density '0'
option mu_beamformer '1'
手动固定信道和频宽，WIFI5大概就这样。改完保存重启就行了。具体看自己机子

这issues咋自己完成了？没见大佬修复啊

Hirro029, 2024-11-23:
试了一下，好了。
我的是少了
option band '5g'
option channel '44'
这两行，其他的都有。
还有就是感觉是新的UI的问题，在保存的时候好像就把默认的这两行删除了。
应该主要还是
option band '5g'
这行的问题
[image]

ashaaf, 2024-11-24:
咦，拉了最新版本后还是一样问题呀，怎么关了

YEYEv50, 2024-11-24:
> 咦，拉了最新版本后还是一样问题呀，怎么关了

手动编辑配置文件吧，应该是LUCI的问题。参数不齐

YEYEv50, 2024-11-24:
> 试了一下，好了。 我的是少了 option band '5g' option channel '44' 这两行，其他的都有。 还有就是感觉是新的UI的问题，在保存的时候好像就把默认的这两行删除了。 应该主要还是 option band '5g' 这行的问题 ![1](https://private-user-images.githubusercontent.com/134217849/389180267-04773580-f686-4685-a82a-d5f0485a74f2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI0NTc1ODIsIm5iZiI6MTczMjQ1NzI4MiwicGF0aCI6Ii8xMzQyMTc4NDkvMzg5MTgwMjY3LTA0NzczNTgwLWY2ODYtNDY4NS1hODJhLWQ1ZjA0ODVhNzRmMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNFQxNDA4MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MzRhZjIzMjE1OWMxOGQzNzI0YzcyYjE2NzQwMzYzNTBiOWIxNDRmOGRiYjQ4N2RkNGM1YWIwN2RiNjBjNWE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.R-9o1lU8RlEHPJE_unzIz8MIhaeluWIpksg71AxVH5w)

应该是LUCI的问题了，我拿IMM的23.05.4编译的固件刷入完全正常。

binext, 2024-12-01:
> 应该是LUCI的问题了，我拿IMM的23.05.4编译的固件刷入完全正常。

难怪，LUCI的问题，我说怎么默认设置正常，只要一编辑5G就无线未关联，无法启用了。我说怎么拉了最新的编译还一直这样。

YEYEv50, 2024-12-03:
> > > 试了一下，好了。 我的是少了 option band '5g' option channel '44' 这两行，其他的都有。 还有就是感觉是新的UI的问题，在保存的时候好像就把默认的这两行删除了。 应该主要还是 option band '5g' 这行的问题 ![1](https://private-user-images.githubusercontent.com/134217849/389180267-04773580-f686-4685-a82a-d5f0485a74f2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI0NTc1ODIsIm5iZiI6MTczMjQ1NzI4MiwicGF0aCI6Ii8xMzQyMTc4NDkvMzg5MTgwMjY3LTA0NzczNTgwLWY2ODYtNDY4NS1hODJhLWQ1ZjA0ODVhNzRmMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNFQxNDA4MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MzRhZjIzMjE1OWMxOGQzNzI0YzcyYjE2NzQwMzYzNTBiOWIxNDRmOGRiYjQ4N2RkNGM1YWIwN2RiNjBjNWE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.R-9o1lU8RlEHPJE_unzIz8MIhaeluWIpksg71AxVH5w)
> > 
> > 
> > 应该是LUCI的问题了，我拿IMM的23.05.4编译的固件刷入完全正常。
> 
> 请问，IMM的23.05.4，是指哪个？我也遇到这个问题了，5G配置页面AC模式没法选择信道，N模式正常，一配置AC模式就启动不了。

看上面的配置参考，手动编辑WIFI的配置吧。