coolsnowwolf/lede #12633: DDNS无法获取IPV6地址

qqsir-dev, 2024-11-01:
### 详细叙述

DDNS里面尝试了各种网络，接口的设置，始终无法获取IPV6地址。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86软路由

### 详细日志

**以下来自系统日志：**
Fri Nov  1 09:20:06 2024 user.warn ddns-scripts[17934]: AliDDNS: NO valid IP found
Fri Nov  1 09:20:06 2024 user.warn ddns-scripts[17934]: AliDDNS: Get registered/public IP for 'homev6.bmwlive.club' failed - retry 1/0 in 60 seconds
Fri Nov  1 09:21:06 2024 user.warn ddns-scripts[17934]: AliDDNS: NO valid IP found
Fri Nov  1 09:21:06 2024 user.warn ddns-scripts[17934]: AliDDNS: Get registered/public IP for 'homev6.bmwlive.club' failed - retry 2/0 in 60 seconds
Fri Nov  1 09:22:06 2024 user.warn ddns-scripts[17934]: AliDDNS: NO valid IP found
Fri Nov  1 09:22:06 2024 user.warn ddns-scripts[17934]: AliDDNS: Get registered/public IP for 'homev6.bmwlive.club' failed - retry 3/0 in 60 seconds
Fri Nov  1 09:23:03 2024 user.notice ddns-scripts[10746]: AliDDNS: PID '10746' started at 2024-11-01 09:23
Fri Nov  1 09:23:03 2024 user.notice ddns-scripts[17934]: AliDDNS: PID '17934' terminated by 'SIGTERM' at 2024-11-01 09:23
Fri Nov  1 09:23:03 2024 user.warn ddns-scripts[10746]: AliDDNS: NO valid IP found
Fri Nov  1 09:23:03 2024 user.warn ddns-scripts[10746]: AliDDNS: Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
Fri Nov  1 09:23:03 2024 user.err ddns-scripts[10746]: AliDDNS: ip Error: '1'
Fri Nov  1 09:23:03 2024 user.warn ddns-scripts[10746]: AliDDNS: Get current IP via 'interface' failed - retry 1/0 in 60 seconds
Fri Nov  1 09:24:03 2024 user.err ddns-scripts[10746]: AliDDNS: ip Error: '1'
Fri Nov  1 09:24:04 2024 user.warn ddns-scripts[10746]: AliDDNS: Get current IP via 'interface' failed - retry 2/0 in 60 seconds
Fri Nov  1 09:25:04 2024 user.err ddns-scripts[10746]: AliDDNS: ip Error: '1'
Fri Nov  1 09:25:04 2024 user.warn ddns-scripts[10746]: AliDDNS: Get current IP via 'interface' failed - retry 3/0 in 60 seconds
Fri Nov  1 09:26:02 2024 user.notice ddns-scripts[13013]: AliDDNS: PID '13013' started at 2024-11-01 09:26
Fri Nov  1 09:26:02 2024 user.notice ddns-scripts[10746]: AliDDNS: PID '10746' terminated by 'SIGTERM' at 2024-11-01 09:26
Fri Nov  1 09:26:02 2024 user.warn ddns-scripts[13013]: AliDDNS: NO valid IP found
Fri Nov  1 09:26:02 2024 user.warn ddns-scripts[13013]: AliDDNS: Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''

**以下来自DDNS的日志：**
 092602       : ************ ************** ************** **************
 092602  note : PID '13013' started at 2024-11-01 09:26
 092602       : ddns version  : 2.8.2-49
 092602       : uci configuration:
ddns.AliDDNS.check_unit='minutes'
ddns.AliDDNS.domain='homev6.bmwlive.club'
ddns.AliDDNS.enabled='1'
ddns.AliDDNS.force_unit='minutes'
ddns.AliDDNS.interface='pppoe-wan'
ddns.AliDDNS.ip_interface='pppoe-wan'
ddns.AliDDNS.ip_source='interface'
ddns.AliDDNS.lookup_host='homev6.bmwlive.club'
ddns.AliDDNS.password='***PW***'
ddns.AliDDNS.retry_unit='seconds'
ddns.AliDDNS.service_name='aliyun.com'
ddns.AliDDNS.use_ipv6='1'
ddns.AliDDNS.use_syslog='2'
ddns.AliDDNS.username='LTAIHiwKt52WZmKg'
ddns.AliDDNS=service
 092602       : verbose mode  : 0 - run normal, NO console output
 092602       : check interval: 600 seconds
 092602       : force interval: 4320 seconds
 092602       : retry interval: 60 seconds
 092602       : retry max count : 0 times
 092602       : 'SIGTERM' was send to old process
 092602  note : PID '10746' terminated by 'SIGTERM' at 2024-11-01 09:26
 092602       : last update: 2024-11-01 09:10
 092602       : Detect registered/public IP
 092602       : #> /usr/bin/nslookup homev6.bmwlive.club  >/var/run/ddns/AliDDNS.dat 2>/var/run/ddns/AliDDNS.err
 092602  WARN : NO valid IP found
 092602  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
 092602  info : Starting main loop at 2024-11-01 09:26
 092603       : Detect current IP on 'interface'
 092603       : #> ip -o addr show dev pppoe-wan scope global >/var/run/ddns/AliDDNS.dat 2>/var/run/ddns/AliDDNS.err
 092603       : Current IP '2408:8206:1860:739d:458b:1ab4:b90e:4dfe' detected on interface 'pppoe-wan'
 092603       : Update needed - L: '2408:8206:1860:739d:458b:1ab4:b90e:4dfe' <> R: 'invalid'
 092603       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 092603       : #> wget --no-hsts -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=homev6.bmwlive.club&DomainName=bmwlive.club&Line=default&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAIHiwKt52WZmKg&SignatureMethod=HMAC-SHA1&Timestamp=2024-11-01T01%3A26%3A03Z&SignatureVersion=1.0&SignatureNonce=559ea1ea-f05f-49bf-8af2-2b0f45a5de15&Signature=ymocTO%2BKuQPd3ZAE5ImROz%2FdTas%3D'
 092603       : 

: {"TotalCount":1,"PageSize":20,"RequestId":"1D4ACC69-6AFB-58DC-9DB9-99A1A2C47CA3","DomainRecords":{"Record":[{"Status":"ENABLE","Line":"default","RR":"homev6","Locked":false,"Type":"AAAA","DomainName":"bmwlive.club","Value":"2408:8207:1866:5d30:0000:0000:0000:0001","RecordId":"839875231472941056","TTL":600,"Weight":1}]},"PageNumber":1}
 092603       : 

ID: 839875231472941056, 

: AAAA
 092603       : 
 092606       : #> wget --no-hsts -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=UpdateDomainRecord&RecordId=839875231472941056&RR=homev6&Type=AAAA&Value=2408%3A8206%3A1860%3A739d%3A458b%3A1ab4%3Ab90e%3A4dfe&Format=JSON&TTL=600&Version=2015-01-09&AccessKeyId=LTAIHiwKt52WZmKg&SignatureMethod=HMAC-SHA1&Timestamp=2024-11-01T01%3A26%3A06Z&SignatureVersion=1.0&SignatureNonce=0924455b-fdb8-418b-9633-df69acd0eef3&Signature=GG61RFuCEcJi4ejaylsuQLVbeqI%3D'
 092606       : 
 092606  info : Update successful - IP '2408:8206:1860:739d:458b:1ab4:b90e:4dfe' send
 092606  info : Forced update successful - IP: '2408:8206:1860:739d:458b:1ab4:b90e:4dfe' send
 092606       : Waiting 600 seconds (Check Interval)

qqsir-dev, 2024-11-01:
DDNS日志貌似是成功的，但是在DDNS里面显示是没有获取到
[image]

qingtian110, 2024-11-01:
DDNS 高级设置

**IP 地址来源：** `URL`
**用于检测的 URL：** `https://6.ipw.cn`

完事

coolsnowwolf, 2024-11-01:
homev6.bmwlive.club 这个地址在你那边被墙了，悲剧

qqsir-dev, 2024-11-01:
> homev6.bmwlive.club 这个地址在你那边被墙了，悲剧

那这个怎么解决？

closed 2024-11-02 by qqsir-dev