coolsnowwolf/lede #12635: 6.1.112内核无法安装Wireguard接口

rex5652, 2024-11-01:
### 详细叙述

截止到 bd85b63920176e009993373ba76936da421f5da2 ，
树莓派4B用的是6.1.112版本的内核，无法支持Wireguard接口，兼容有问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

树莓派4B

### 详细日志

[image]

xingshu1990, 2024-11-02:
不，你这个是编译的时候 内核没设定为6.1，编译出来的ipk还是6.6内核的。
我的理解：
lede/target/linux/bcm(不知道树莓派4B选择什么)
打开Makefile文件修改，修改后记得保存。
KERNEL_PATCHVER:=6.1                    #5.15修改成5.4

但是不知道是不是还要改include/kernel-version.mk（因为这里有提到https://www.openwrt.pro/post-638.html）

内核改变，编译前建议清理下：
make dirclean

rex5652, 2024-11-02:
op官方的master分支已经对树莓派4b  bcm7211提供了6.6的内核，其他架构也有6.6的内核。只要都同步到6.6就行了

aiamadeus, 2024-11-02:
跟内核版本有毛关系，自己编译的固件，内核模块不能安装别人编的

closed 2024-11-20 by aiamadeus