coolsnowwolf/lede #12640: luci23.05下怎么修改默认主题？

fxxccp, 2024-11-03:
### 详细叙述

旧版luci修改默认主题是在feeds/luci/collections/luci/Makefile文件下，而luci23.05下的同名文件中没有找到修改的字串。请问休班luci怎么修改默认主题？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

shing-lee, 2024-11-03:
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci/Makefile
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci-light/Makefile
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci-nginx/Makefile
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci-ssl-nginx/Makefile

真正生效的是第2条，但是建议4条一起替换。

fxxccp, 2024-11-05:
> sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci/Makefile

感谢指点迷津。不过第一条貌似无效，文件中并没有luci-theme-bootstrap这一字串。

shing-lee, 2024-11-05:
> > sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' 源/luci/collections/luci/Makefile
> 
> 感谢指点迷津。不过第一条貌似无效，文件中并没有luci-theme-bootstrap这一字串。

第一条是用来改老版本luci的，因为不同的编译环境替换的文件不一样，所以我就干脆全部都换，反正没有的也替换不了，哈哈。

allok1985, 2024-11-07:
我编译含Design主题，按上面的修改升级就换层Design了，有其他办法吗？

closed 2024-11-08 by coolsnowwolf