coolsnowwolf/lede #12648: 麻烦修复一下lame

xuanranran, 2024-11-06:
### 详细叙述

make[3]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make  all-recursive
make[4]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
Making all in mpglib
make[5]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
Makefile:426: .deps/tabinit.Plo: No such file or directory
make[5]: *** No rule to make target '.deps/tabinit.Plo'.  Stop.
make[5]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
make[4]: *** [Makefile:452: all-recursive] Error 1
make[4]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[3]: *** [Makefile:381: all] Error 2
make[3]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[2]: *** [Makefile:96: /home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/.built] Error 2
make[2]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/customfeeds/packages/sound/lame'
time: package/feeds/packages/lame/compile#0.08#0.00#0.08
    ERROR: package/feeds/packages/lame failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/lame/compile] Error 1

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

make[3]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make  all-recursive
make[4]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
Making all in mpglib
make[5]: Entering directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
Makefile:426: .deps/tabinit.Plo: No such file or directory
make[5]: *** No rule to make target '.deps/tabinit.Plo'.  Stop.
make[5]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
make[4]: *** [Makefile:452: all-recursive] Error 1
make[4]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[3]: *** [Makefile:381: all] Error 2
make[3]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[2]: *** [Makefile:96: /home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/lame-3.100/.built] Error 2
make[2]: Leaving directory '/home/ranran/actions-runner/_work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/customfeeds/packages/sound/lame'
time: package/feeds/packages/lame/compile#0.08#0.00#0.08
    ERROR: package/feeds/packages/lame failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/lame/compile] Error 1

MilesPoupart, 2024-11-07:
这个问题我也碰到过，还蛮奇怪的。我在一台机子上的WSL可以编译lame成功，但是原样复制WSL的vhdx到另一台机器就会编译lame失败，详细日志是：
```
make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/customfeeds/packages/sound/lame'
rm -f /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/.built
touch /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100=lame-3.100 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100=lame-3.100 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make -j1 -C /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100=lame-3.100 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[3]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make  all-recursive
make[4]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
Making all in mpglib
Makefile:426: .deps/tabinit.Plo: No such file or directory
make[5]: *** No rule to make target '.deps/tabinit.Plo'.  Stop.
make[5]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
make[5]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/mpglib'
make[4]: *** [Makefile:452: all-recursive] Error 1
make[4]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[3]: *** [Makefile:381: all] Error 2
make[3]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100'
make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/customfeeds/packages/sound/lame'
make[2]: *** [Makefile:96: /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/lame-3.100/.built] Error 2
time: package/feeds/packages/lame/compile#0.08#0.00#0.08
    ERROR: package/feeds/packages/lame failed to build.
```
需要手动进入lame的编译目录执行autoreconf -i命令之后才会生成相应的.deps中的文件，然后才能make编译成功，粗看了一下makefile里头是有PKG_FIXUP := autoreconf的，而且另一台能成功的机子也不需要这样手动干预，很奇怪。

xuanranran, 2024-11-07:
> make[5]: *** No rule to make target '.deps/tabinit.Plo'. Stop.

应该说他是要创建`mkdir -p .deps` 文件夹 再进行 `autoreconf -fi` 才成功

closed 2024-11-07 by coolsnowwolf

MilesPoupart, 2024-11-07:
？怎么就completed了 俺还想蹲一个排查方案

xuanranran, 2024-11-07:
> ？怎么就completed了 俺还想蹲一个排查方案

我也挺纳闷的 可能大雕不想修吧

MilesPoupart, 2024-11-07:
看了下sourceforge的lame-3.100源码里头自带几个 automake 1.15.1 生成的Makefile.in 失败的机子好像没有对这些makefile.in重新生成

MilesPoupart, 2024-11-07:
make package/lame/compile -j1 V=s编译了一下看了下失败的log里会有这么一条：
```
configure.in:425: error: possibly undefined macro: AM_ICONV
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.configure.in:425: error: possibly undefined macro: AM_ICONV
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
```

MilesPoupart, 2024-11-07:
参考https://github.com/openwrt/packages/issues/20677#issuecomment-1483774965修改下 我这边自己试了试成功了 周末我看看提交个PR

xuanranran, 2024-11-08:
> 己试了试可以编译成功了 周

我待会试试看看 替换packages的lame

xuanranran, 2024-11-08:
> 参考[openwrt/packages#20677 (comment)](https://github.com/openwrt/packages/issues/20677#issuecomment-1483774965) 修改下[MilesPoupart/packages@ecb5887](https://github.com/MilesPoupart/packages/commit/ecb58874023915deb8bab02148d993409235d177) 我这边自己试了试可以编译成功了 周末我看看提交个PR

刚刚编译可以诶，有空可以提交一下PR