coolsnowwolf/lede #12827: [[Watch~Live]] San Jacinto vs Newbury Park Live 08 November 2024

fuiplasciX, 2024-11-08:
### 详细叙述

Game information reflects the most current available data from NFHS. To see how NFHS collects their information, click here.

Follow high school football this season on the NFHS Network. Keep tabs on your family or alma mater and tune in.

Central (Miami, Fla.), a nine-time state champion, was ordered Monday by the Florida governing body to forfeit all nine games it had played this season for using an ineligible player, according to a story in the Palm Beach Post.

The Rockets used a fifth-year high school student-athlete in compiling an 8-1 mark, according to the Florida High School Athletic Association. The forfeits drop the Rockets to 0-9 and all but assures Miami Central will miss the playoffs when the FHSAA releases the brackets later this week.

The Palm Beach Post said Miami Central self-reported the violation, which stated the ineligible player "entered the 9th grade for the first time during the 2020-2021 school year."

FHSAA Bylaw 9.5.1 states that a student is limited to eight consecutive semesters of eligibility beginning with the semester they begin ninth grade for the first time. After eight consecutive semesters, the student is permanently ineligible.
Nine-time state champion Miami Central was ordered on Monday to forfeit all nine of its games from the 2024 season by the FHSAA for using an ineligible player. The Rockets drop to 0-9 and likely will miss the postseason when the brackets are released later this week. (File photo: Laura Martin)

Nine-time state champion Miami Central was ordered on Monday to forfeit all nine of its games from the 2024 season by the FHSAA for using an ineligible player. The Rockets drop to 0-9 and likely will miss the postseason when the brackets are released later this week. (File photo: Laura Martin)

The Rockets began the season at No. 15 in the MaxPreps Top 25, but fell out of the rankings after a season-opening loss to Lakeland and hadn't climbed back in. They sat at No. 27 in the expanded rankings, but are likely to fall out of the top 100 following the news.

They were also the top-ranked team in the MaxPreps computer rankings in Class 3A and were likely going to be the top seed in the region's postseason bracket.

The news about Miami Central was the latest in a string of forfeits stemming from FHSAA sanctions.

Norland (Miami, Fla.) was forced to forfeit its 33-27 win against American Heritage on Oct. 27 for playing an ineligible player.

But the biggest punishment until the Miami Central story broke had been the FHSAA dropping the hammer on The First Academy (Orlando, Fla.) over the weekend.

The Class 1A team was ordered to forfeit all of its games, fined $36,000 and banned from postseason for this year and 2025. The punishment stems from an investigation that found the school used ineligible players, provided impermissible benefits to players, allowed athletes to be involved in football activities without being formally enrolled and improperly communicated with student-athletes prior to their enrollment at The First Academy. The school has 10 days to appeal the sanctions.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

SDFGHJ

### 详细日志

DFGHJK

DXFGHJK

DFGHJ
DFGHJK
SDFT

closed 2024-11-08 by coolsnowwolf

daeypaover, 2024-11-08:
csfdsf

borchiposao99, 2024-11-08:
com to 12