coolsnowwolf/lede #12883: 树莓派4b刷openwrt，正常启动ssr+等各种代理，但是无法ping通google怎么解决，求助各位大佬愿奉上一杯奶茶以表感谢（抱拳

Jiayiiiio, 2024-11-10:
### 详细叙述

描述：【小白一个，有些许linux经验，对网络环境配置和openwrt使用比较薄弱，求助各位大佬愿奉上一杯奶茶以表感谢（抱拳（抱拳 】
使用目的：树莓派4b刷openwrt做wifi路由器，方便科学上网和使用docker，方便出差场景中使用熟悉的网络环境。
硬件描述：树莓派4b刷openwrt固件，机器本身只有一个lan口无其他网口，目前机器直连光猫网关。路由器lan口静态配置192.168.1.xx。
问题描述：正常启动ssr+等各种代理，无法在ssh中ping通www.google.com，其他设备连接WiFi也无法科学上网。
[image]

1. ssr+日志如下：

```makefile
2024-11-10 10:49:35: -----------end------------
2024-11-10 10:49:35: Apple 域名中国大陆 CDN 的 优化规则加载完毕。
2024-11-10 10:49:35: Apple 域名中国大陆 CDN 的 优化规则正在加载。
2024-11-10 10:49:26: Main node:Welcome to trojan 1.16.0 , 4 Threads Started!
2024-11-10 10:49:26: ----------start------------
2024-11-10 10:13:56: -----------end------------
2024-11-10 10:13:55: Apple 域名中国大陆 CDN 的 优化规则加载完毕。
2024-11-10 10:13:55: Apple 域名中国大陆 CDN 的 优化规则正在加载。
2024-11-10 10:13:49: Main node:Welcome to trojan 1.16.0 , 4 Threads Started!
2024-11-10 10:13:49: ----------start------------
2024-11-10 10:09:09: -----------end------------
2024-11-10 10:09:09: Apple 域名中国大陆 CDN 的 优化规则加载完毕。
2024-11-10 10:09:09: Apple 域名中国大陆 CDN 的 优化规则正在加载。
2024-11-10 10:08:58: Main node:Welcome to trojan 1.16.0 , 4 Threads Started!
2024-11-10 10:08:58: ----------start------------
2024-11-10 09:58:12: -----------end------------
2024-11-10 09:58:11: Apple 域名中国大陆 CDN 的 优化规则加载完毕。
2024-11-10 09:58:11: Apple 域名中国大陆 CDN 的 优化规则正在加载。
2024-11-10 09:58:03: Main node:Welcome to trojan 1.16.0 , 4 Threads Started!
2024-11-10 09:58:03: ----------start------------
2024-11-10 04:01:13: -----------end------------
2024-11-10 04:01:12: Apple 域名中国大陆 CDN 的 优化规则加载完毕。
2024-11-10 04:01:12: Apple 域名中国大陆 CDN 的 优化规则正在加载。
2024-11-10 04:01:06: Main node:Welcome to trojan 1.16.0 , 4 Threads Started!
2024-11-10 04:01:06: ----------start------------
2024-11-10 04:01:03 订阅更新成功
2024-11-10 04:01:03 新增节点数量: 0 删除节点数量: 0
2024-11-10 04:01:03 维持当前主服务器节点。
2024-11-10 04:01:03 成功解析节点数量: 30
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IPV6|新加坡1|x0.1
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IPV6|美国1|x0.1
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IEPL1|香港1|x3
2024-11-10 04:01:03 丢弃无效节点: trojan 节点, IEPL2|马来西亚|原生IP|x3
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IEPL2|俄罗斯|原生IP|x3
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IEPL2|瑞典|原生IPx3
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IEPL2|英国|原生IP|x3
2024-11-10 04:01:03 丢弃无效节点: trojan 节点, IEPL1|土耳其|原生IP|x3
2024-11-10 04:01:03 丢弃无效节点: ss 节点, IEPL1|阿根廷1|x3
2024-11-10 04:01:03 丢弃无效节点: trojan 节点, CM2|菲律宾1|原生IP|x1
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, CM2|英国1|原生IP|x1
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, CM1|英国1|原生IP|x1
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, CM2|香港1|x1
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, CM1|香港1|x1
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, 套餐到期：2024-11-21
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, 距离下次重置剩余：11 天
2024-11-10 04:01:02 丢弃无效节点: trojan 节点, 剩余流量：453.64 GB
2024-11-10 04:00:58 服务正在暂停
2024-11-10 04:00:54 你已经是最新数据，无需更新！
2024-11-10 04:00:32 正在更新【国内IP段】数据库
2024-11-10 04:00:32 你已经是最新数据，无需更新！
2024-11-10 04:00:00 正在更新【GFW列表】数据库
``` 

2. ssr状态：

[image]

### 重复 issue

[... 385 lines cut ...]
[  325.842025] br-lan: port 1(eth0) entered blocking state
[  325.847297] br-lan: port 1(eth0) entered forwarding state
[  326.565868] br-lan: port 1(eth0) entered disabled state
[  328.236478] br-lan: port 2(wlan0) entered blocking state
[  328.241826] br-lan: port 2(wlan0) entered disabled state
[  328.247324] device wlan0 entered promiscuous mode
[  328.966011] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  328.974265] br-lan: port 1(eth0) entered blocking state
[  328.979535] br-lan: port 1(eth0) entered forwarding state
[  330.803315] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  330.809905] br-lan: port 2(wlan0) entered blocking state
[  330.815233] br-lan: port 2(wlan0) entered forwarding state
[  392.405673] bcmgenet fd580000.ethernet eth0: Link is Down
[  392.411428] br-lan: port 1(eth0) entered disabled state
[  416.325971] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  416.334036] br-lan: port 1(eth0) entered blocking state
[  416.339289] br-lan: port 1(eth0) entered forwarding state
[  448.543149] device wlan0 left promiscuous mode
[  448.547784] br-lan: port 2(wlan0) entered disabled state
[  450.979802] br-lan: port 2(wlan0) entered blocking state
[  450.985138] br-lan: port 2(wlan0) entered disabled state
[  450.990579] device wlan0 entered promiscuous mode
[  453.714349] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  453.720968] br-lan: port 2(wlan0) entered blocking state
[  453.726315] br-lan: port 2(wlan0) entered forwarding state

bgylde, 2024-11-10:
没用过这个插件，不过我的理解是ssr属于应用层协议，不应该用icmp去测试。可以用curl -v "https://www.google.com"测试输出情况。

xingshu1990, 2024-11-10:
不能ping www.google.com的问题，上面这楼已经解答了。

树莓派系列的WIFI功能，就算了吧，包括但不限于3B+  4B 都是很差。
更推荐树莓派装SSR 和DNS服务，然后硬路由的网关和DNS都设定为当前路由器的IP。

[image]

xingshu1990, 2024-11-10:
[image]

Jiayiiiio, 2024-11-10:
> 没用过这个插件，不过我的理解是ssr属于应用层协议，不应该用icmp去测试。可以用curl -v "[https://www.google.com"测试输出情况。](https://www.google.com%22%E6%B5%8B%E8%AF%95%E8%BE%93%E5%87%BA%E6%83%85%E5%86%B5%E3%80%82)

非常感谢解答，curl -v https://www.google.com结果如下：
root@BleachWrt:~# curl -v  https://google.com
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
> GET / HTTP/2
> Host: google.com
> User-Agent: curl/8.6.0
> Accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
< HTTP/2 301 
< location: https://www.google.com/
< content-type: text/html; charset=UTF-8
< content-security-policy-report-only: object-src 'none';base-uri 'self';script-src 'nonce-K3a4nxgfjIZFoQNazx9z4w' 'strict-dynamic' 'report-sample' 'unsafe-eval' 'unsafe-inline' https: http:;report-uri https://csp.withgoogle.com/csp/gws/other-hp
< date: Sun, 10 Nov 2024 08:01:40 GMT
< expires: Tue, 10 Dec 2024 08:01:40 GMT
< cache-control: public, max-age=2592000
< server: gws
< content-length: 220
< x-xss-protection: 0
< x-frame-options: SAMEORIGIN
< alt-svc: h3=":443"; ma=2592000,h3-29=":443"; ma=2592000
< 
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="https://www.google.com/">here</A>.
</BODY></HTML>

Jiayiiiio, 2024-11-10:
> 不能ping [www.google.com](http://www.google.com) 的问题，上面这楼已经解答了。
> 
> 树莓派系列的WIFI功能，就算了吧，包括但不限于3B+ 4B 都是很差。
> 
> 更推荐树莓派装SSR 和DNS服务，然后硬路由的网关和DNS都设定为当前路由器的IP。这个是办法之一： 主路由开启wifi功能，树莓派的wifi，关不关都没关系，反正只要不连这个WIFI。
> 
> ![image](https://private-user-images.githubusercontent.com/18671173/384664288-9765f7f9-408d-42c5-ad64-66f90b3985f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyMjU1NzMsIm5iZiI6MTczMTIyNTI3MywicGF0aCI6Ii8xODY3MTE3My8zODQ2NjQyODgtOTc2NWY3ZjktNDA4ZC00MmM1LWFkNjQtNjZmOTBiMzk4NWYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEwVDA3NTQzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3NmU3ZjllZDFkOWRjZDc5OTc1NzRiNGE1YzU2M2ZlMTE5NmY4YmMzM2FiMzM2NTJkNGViMDg0OWFmMDlmMzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QaxGtqFGsRetIxK37eHFqvVkrgklRygCocfNJKtFZdU)

非常感谢回答，自己构想中树莓派路由未来带着酒店插网线临时使用。您描述的这些操作我已经调整但是还是没有办法成功科学上网

bgylde, 2024-11-10:
> > 没用过这个插件，不过我的理解是ssr属于应用层协议，不应该用icmp去测试。可以用curl -v "[https://www.google.com"测试输出情况。](https://www.google.com%22%E6%B5%8B%E8%AF%95%E8%BE%93%E5%87%BA%E6%83%85%E5%86%B5%E3%80%82)
> 
> 非常感谢解答，curl -v [https://www.google.com结果如下：](https://www.google.com%E7%BB%93%E6%9E%9C%E5%A6%82%E4%B8%8B%EF%BC%9A) root@BleachWrt:~# curl -v https://google.com
> 
> * TLSv1.3 (OUT), TLS handshake, Client hello (1):
> * TLSv1.3 (IN), TLS handshake, Server hello (2):
> * TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
> * TLSv1.3 (IN), TLS handshake, Certificate (11):
> * TLSv1.3 (IN), TLS handshake, CERT verify (15):
> * TLSv1.3 (IN), TLS handshake, Finished (20):
> * TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
> * TLSv1.3 (OUT), TLS handshake, Finished (20):
> 
> > GET / HTTP/2
> > Host: google.com
> > User-Agent: curl/8.6.0
> > Accept: _/_
> 
> * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
> * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
>   < HTTP/2 301
>   < location: https://www.google.com/
>   < content-type: text/html; charset=UTF-8
>   < content-security-policy-report-only: object-src 'none';base-uri 'self';script-src 'nonce-K3a4nxgfjIZFoQNazx9z4w' 'strict-dynamic' 'report-sample' 'unsafe-eval' 'unsafe-inline' https: http:;report-uri https://csp.withgoogle.com/csp/gws/other-hp
>   < date: Sun, 10 Nov 2024 08:01:40 GMT
>   < expires: Tue, 10 Dec 2024 08:01:40 GMT
>   < cache-control: public, max-age=2592000
>   < server: gws
>   < content-length: 220
>   < x-xss-protection: 0
>   < x-frame-options: SAMEORIGIN
>   < alt-svc: h3=":443"; ma=2592000,h3-29=":443"; ma=2592000
>   <
> 
> <TITLE>301 Moved</TITLE>
> # 301 Moved
> The document has moved [here](https://www.google.com/).

看结果google已经返回数据，301重定向到 https://www.google.com/ ，证明路由器配置没有问题。可以直接在电脑上测试浏览器是否可以访问，如果失败的话，就需要看看电脑的网络配置情况。

closed 2024-11-20 by coolsnowwolf