coolsnowwolf/lede #12987: 无法更新安装插件,这个问题要如何解决。之前没有遇到过

Ray2023m, 2024-11-13:
### 详细叙述

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
*** Failed to download the package list from https://istore.linkease.com/repo/all/compat/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony

xingshu1990, 2024-11-13:
mirrors.cloud.tencent.com 腾讯的连接没用了 或者近期失效。

换成https://downloads.openwrt.org/releases/的试试

Ray2023m, 2024-11-15:
这个mirrors.cloud.tencent.com 是可以正常访问的，但是一直报错的。专门进去看了，里面根本就没有[Packages.gz]这个文件(https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz)
> ## 2024-11-14更新：
> mirrors.cloud.tencent.com能正常访问了。
> mirrors.cloud.tencent.com 腾讯的连接没用了 或者近期失效。 换成[https://downloads.openwrt.org/releases/的试试](https://downloads.openwrt.org/releases/%E7%9A%84%E8%AF%95%E8%AF%95)

Ray2023m, 2024-11-15:
自己编译的。x86.内核也是最新的。但是还是出现同样错误。尝试更换openwrt源，可以刷新。但是无法安装。内核太新

closed 2024-11-18 by coolsnowwolf